tesseract-robotics/tesseract_ros #5: tesseract_ros_examples nodes always crash at startup

bgd29, 2020-03-24:
The tesseract_ros_examples nodes crash while trying to roslaunch them. I've tried to use gdb and I get a SIGSEGV error:
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff0a6735f in ?? () from /usr/lib/x86_64-linux-gnu/libjsoncpp.so.1

I'm using ROS1: Melodic on Ubuntu 18.04, gcc version is 7.5. Maybe there's an ABI mismatch causing this?

mpowelson, 2020-04-02:
Sorry no one responded to this. Are you still having this issue?

bgd29, 2020-04-03:
Yes, still dealing with this, not sure what's going on, I'll just try with a clean Ubuntu installation. Is there anything do you think I should try before that?

mpowelson, 2020-04-03:
I think we've had issues with libjsoncpp. Have you tried installing it from source?

Is this what you ran into @Levi-Armstrong @drchrislewis

Levi-Armstrong, 2020-04-03:
I had a similar issue. We had to build jsoncpp from source on some machines and comment out two lines of code. See this [issue](https://github.com/open-source-parsers/jsoncpp/issues/1107).

bgd29, 2020-04-07:
Fantastic, that solved the issue, thank you.

closed 2020-04-07 by bgd29

Levi-Armstrong, 2020-04-07:
Great! I am still not sure why some keep running into this.