tesseract-robotics/tesseract_ros #6: Issue building OMPL examples

hzyjerry, 2020-03-28:
Hi, I'm using tesseract to learn about motion planning and building my codebase. I want to thank you for the wonderful effort.

I wanted to ask about building the latest tesseract examples with OMPL. I tried OMPL version 1.5.0, 1.4.2 and 1.4.0, and have run into the similar build issue below. I wonder if this is due to some version mismatch? Thanks!

```
/home/jerrhe/ws_tesseract/devel/lib/libtesseract_motion_planners_ompl.so: undefined reference to `ompl::geometric::RRT::RRT(std::shared_ptr<ompl::base::SpaceInformation> const&, bool)'
/home/jerrhe/ws_tesseract/devel/lib/libtesseract_motion_planners_ompl.so: undefined reference to `vtable for ompl::base::ProjectedStateSpace'
/home/jerrhe/ws_tesseract/devel/.private/tesseract_ros_examples/lib/libtesseract_ros_examples_glass_upright_ompl_example.so: undefined reference to `ompl::base::Constraint::isSatisfied(Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> > const&) const'
/home/jerrhe/ws_tesseract/devel/.private/tesseract_ros_examples/lib/libtesseract_ros_examples_glass_upright_ompl_example.so: undefined reference to `ompl::base::Constraint::distance(Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> > const&) const'
/home/jerrhe/ws_tesseract/devel/.private/tesseract_ros_examples/lib/libtesseract_ros_examples_glass_upright_ompl_example.so: undefined reference to `ompl::base::Constraint::project(Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1> >) const'
/home/jerrhe/ws_tesseract/devel/lib/libtesseract_motion_planners_ompl.so: undefined reference to `vtable for ompl::base::ConstrainedStateSpace'
/home/jerrhe/ws_tesseract/devel/.private/tesseract_ros_examples/lib/libtesseract_ros_examples_glass_upright_ompl_example.so: undefined reference to `vtable for ompl::base::Constraint'
/home/jerrhe/ws_tesseract/devel/lib/libtesseract_motion_planners_ompl.so: undefined reference to `ompl::base::Constraint::isSatisfied(ompl::base::State const*) const'
/home/jerrhe/ws_tesseract/devel/lib/libtesseract_motion_planners_ompl.so: undefined reference to `ompl::base::ConstrainedStateSpace::ConstrainedStateSpace(std::shared_ptr<ompl::base::StateSpace> const&, std::shared_ptr<ompl::base::Constraint> const&)'
/home/jerrhe/ws_tesseract/devel/lib/libtesseract_motion_planners_ompl.so: undefined reference to `vtable for ompl::base::WrapperStateSpace'
/home/jerrhe/ws_tesseract/devel/lib/libtesseract_motion_planners_ompl.so: undefined reference to `ompl::base::ConstrainedStateSpace::setSpaceInformation(ompl::base::SpaceInformation*)'
/home/jerrhe/ws_tesseract/devel/lib/libtesseract_motion_planners_ompl.so: undefined reference to `ompl::base::SpaceInformation::SpaceInformation(std::shared_ptr<ompl::base::StateSpace>)'
/home/jerrhe/ws_tesseract/devel/.private/tesseract_ros_examples/lib/libtesseract_ros_examples_glass_upright_ompl_example.so: undefined reference to `ompl::base::Constraint::jacobian(Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> > const&, Eigen::Ref<Eigen::Matrix<double, -1, -1, 0, -1, -1>, 0, Eigen::OuterStride<-1> >) const'
/home/jerrhe/ws_tesseract/devel/lib/libtesseract_motion_planners_ompl.so: undefined reference to `ompl::geometric::RRTConnect::RRTConnect(std::shared_ptr<ompl::base::SpaceInformation> const&, bool)'
/home/jerrhe/ws_tesseract/devel/.private/tesseract_ros_examples/lib/libtesseract_ros_examples_glass_upright_ompl_example.so: undefined reference to `typeinfo for ompl::base::Constraint'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/jerrhe/ws_tesseract/devel/.private/tesseract_ros_examples/lib/tesseract_ros_examples/tesseract_ros_examples_glass_upright_ompl_example_node] Error 1
make[1]: *** [CMakeFiles/tesseract_ros_examples_glass_upright_ompl_example_node.dir/all] Error 2
make: *** [all] Error 2
```

Levi-Armstrong, 2020-03-28:
Are you on Ubunutu 16.04 or 18.04?

hzyjerry, 2020-03-29:
I’m using 16.04. OMPL is built from its git repo.

Levi-Armstrong, 2020-03-29:
I assume it is finding the system version. Are you building ompl within your workspace?

hzyjerry, 2020-03-29:
You're right! I was able to get it to build after a deep clean of system packages. 

I noticed that the glass_upright_ompl_example finds no valid solution
```bash
[ERROR] [1585464849.705942978]: Failed to find a valid trajector: Failed to find valid solution
```
Is this the expected behavior?

[image]

Levi-Armstrong, 2020-03-29:
The ompl freespace should work but the glass upright does not. There is an issues with how OMPL handles constraints which causes it never find a valid solution. We are currently testing a modified RRTConnect to address the issue. @cbw36 can you post the location of where you are working on the modified RRTConnect to address OMPL constrained motion planning?

Levi-Armstrong, 2021-02-15:
Please reopen if still valid with the recent merge of the command language branch.

closed 2021-02-15 by Levi-Armstrong