tesseract-robotics/tesseract_ros #10: Question about .csv file generation

jianye0428, 2020-04-08:
hello,

i'm running the tessetact_ros_example -- pick_and_place_example and changed the code in pick_and_place_example_node.cpp:

` bool write_to_file = true;`

but when I check the generated .csv files, I found that there is nothing in the file_output_place.csv file. In file_output_pick.csv file I found the information about pick process.

I don't know whether i did something wrong. look forward for help!

thanks in advance

Jian

mpowelson, 2020-04-08:
It looks like there is a copy and paste error there. Where we are defining the output file for the place move we have the name of the package wrong. Compare [Pick](https://github.com/ros-industrial-consortium/tesseract_ros/blob/4849008faebcd64b9dab2a379099f9bf0e9e04bb/tesseract_ros/tesseract_examples/tesseract_ros_examples/src/pick_and_place_example.cpp#L291) to [Place](https://github.com/ros-industrial-consortium/tesseract_ros/blob/4849008faebcd64b9dab2a379099f9bf0e9e04bb/tesseract_ros/tesseract_examples/tesseract_ros_examples/src/pick_and_place_example.cpp#L501). This should be fixed if you want to submit a PR.

jianye0428, 2020-04-08:
i have also noted that the save path is different but I did also find the file_output_place.csv in "pick_and_place " folder and no data in csv. I also just changed the both save path as tesseracr_ros_example but still no data in file_output_place.csv file but file_output_pick.csv file worked.

thanks for reply!

Jian

mpowelson, 2020-04-08:
That's odd. I don't have a workspace with this in it handy to test out. Maybe try printing out the path that it is supposed to save to. We haven't used that write callback in a while, but I don't know why it would be broken. If that doesn't work, try putting a breakpoint in the callback and make sure it is going inside it.

jianye0428, 2020-06-14:
#20 the problem has been solved.

closed 2020-06-14 by jianye0428