tesseract-robotics/tesseract_ros #11: Divergence of optimization procedures in the cart_seat_example

IKapitaniuk, 2020-04-22:
Hi, I was playing around with the tesseract_ros_examples and found out that the optimization procedure diverges after few steps with the following output:
```
[ERROR] approximate merit function got worse (-8.939e-02). (convexification is probably wrong to zeroth order)
[ERROR] [1587587556.563473493]: Hit enter key to step optimization!
[ERROR] approximate merit function got worse (-3.322e+00). (convexification is probably wrong to zeroth order)
[ERROR] [1587587558.510202476]: Hit enter key to step optimization!
[ERROR] approximate merit function got worse (-3.586e+01). (convexification is probably wrong to zeroth order)
[ERROR] [1587587560.336022855]: Hit enter key to step optimization!
[ERROR] approximate merit function got worse (-3.613e+02). (convexification is probably wrong to zeroth order)
[ERROR] [1587587562.616304592]: Hit enter key to step optimization!
[ERROR] approximate merit function got worse (-3.615e+03). (convexification is probably wrong to zeroth order)
[ERROR] [1587587564.562469531]: Hit enter key to step optimization!
[ INFO] [1587587569.391541770]: Place seat #1 planning time: 43.795
[ INFO] [1587587570.043971946]: Place seat #1 trajectory is collision free
[ INFO] [1587587570.121162704]: RViz environment Updated!
```
Are there any problems with the proper adjustments of constraints and penalties, or the geometry of objects is too complicated for the convexification step of the optimization algorithm?

gavanderhoorn, 2020-04-24:
@IKapitaniuk: could you add some more information about *which* particular example this happens with? All of them? A particular one?

Are you using the examples verbatim, or have you changed anything?

Which version of ROS are you using, and how did you install it? Are you using the default solver, or something else?

gavanderhoorn, 2020-04-29:
ping @IKapitaniuk

ykapitanyuk, 2020-04-29:
1) I have checked all the tests and found that behavior only for a particular "car_seat_example" case.

2) Yes, I used a fresh clean installation without any changes.

3) ROS Melodic on Unbuntu 18.04. Installed from standard binary packages without any custom modification. 

I have tested it with BPMPD and Gurobi. The divergence rate(here I mean the growth of the absolute value of the merit function) with Gurobi is slower, but, in general, the behavior is the same.

Levi-Armstrong, 2020-04-29:
I will need to take a look at the example, but it may take me bit to get to this.

gavanderhoorn, 2020-04-30:
Have you ran into any similar problems with the demos/examples @Levi-Armstrong ?

mpowelson, 2020-04-30:
I just tried it. It converged for me, but something odd happened. I found this amusing.

[image]

IKapitaniuk, 2020-04-30:
Sorry, I forgot to mention that I've redirected std output, adding "output=screen" to the corresponding launch file "car_seat_example.launch", to see that it diverges. 
With the standard configuration, this output from a solver is suppressed. 
Also, please note that despite the divergence of the optimization process it still managed to find a collision-free trajectory.

Levi-Armstrong, 2022-04-02:
The examples recently updated add tested so closing. Please reopen if issues still persist.

closed 2022-04-02 by Levi-Armstrong