tesseract-robotics/tesseract_ros #12: Problem using tesseract_monitoring for refreshing the scene in rviz

sduminy, 2020-04-28:
Hi all,
I am having an issue related I think to the tesseract_monitoring behavior, which I don't fully understand.
I want to test a pick and place example I wrote using a library of motion planning based on tesseract. In my example, I create 4 boxes corresponding to a table, a shelf with one object on each of them. Then I call my function to compute the grasping pose before setting it as an eef goal. Then I call a function which attach the box to the eef and then call the function computing the place pose before setting it as new eef goal.
My problem is when I use `environment_monitor_->triggerEnvironmentUpdateEvent(envrionment_monitor_->UPDATE_ENVIRONMENT)` in the function adding the box, it works the first time I run my test but if I try again, then I have warnings saying "Tried to add link (object_1) with same name as existing link" or the same message with the object on the shelf and the object won't attach to the eef and the robot just moves to the place pose without the object in his gripper.
My code is inspired by this example : [scene_graph_example.cpp](https://github.com/ros-industrial-consortium/tesseract_ros/blob/master/tesseract_ros/tesseract_examples/tesseract_ros_examples/src/scene_graph_example.cpp)
However, if I remove the refreshing of the environment in the function adding the box, the scene appears when the robot reaches the grasp pose and every element is where it is supposed to be except 2 additional boxes, which lie at the robot initial position I guess. But after the robot starts moving to his place pose, they disappear and the behavior is as expecting (with the object in the gripper of the robot).
[image]
[image]
[image]
I should add that I am only working with rviz as visualization.

Do you know where does my issue come from and could you help me solve it ?

Thanks for your time,

sduminy, 2020-04-28:
I have a simpler question : should I use `envrionment_monitor_->triggerEnvironmentUpdateEvent()` to display the new objects in the scene ? I have a feeling my issue comes from here.

Levi-Armstrong, 2020-04-29:
Are you using the Environment Monitor as a ROS Node or a library within your node?

If you are using it as a library you just need to get the Environment from the the Environment Monitor using `getEnvironment()` then move thinks around using the methods from the Environment class. Make sure to lock the environment when making the changes leveraging LockedEnvironmentRO and LockedEnvironmentRW. In your case you want to move a joint around.

sduminy, 2020-04-29:
I am using environment monitor as a library within my node.
Should I make every changes to the environment with the pointer from environment monitor to `getEnvironment()` ? 
In my code, I used a pointer from tesseract to `getEnvironment()` for my changes and I did the refresh using a pointer from environment monitoring so I guess my error comes from that ?
I don't really get how to use LockedEnvironmentRO/RW. Could you explain me briefly ?
Thank you for your time,

Levi-Armstrong, 2020-04-29:
If you are using it as a library then I would use the getEnvironment()  and when making changes or reading you must call one of the methods below to perform the correct lock because things are operating in a different threads and can be modified by incoming topic or service messages causing a race condition.

Example:
```
environment_monitor_->lockEnvironmentWrite()
// Add code here which modifies the environment
environment_monitor_->unlockEnvironmentWrite()
```

```
void EnvironmentMonitor::lockEnvironmentRead() { scene_update_mutex_.lock_shared(); }
void EnvironmentMonitor::unlockEnvironmentRead() { scene_update_mutex_.unlock_shared(); }
void EnvironmentMonitor::lockEnvironmentWrite() { scene_update_mutex_.lock(); }
void EnvironmentMonitor::unlockEnvironmentWrite() { scene_update_mutex_.unlock(); }
```

sduminy, 2020-04-30:
This doesn't solve my initial problem of displaying the scene correctly. Should I call `environment_monitor_->triggerEnvironmentUpdateEvent(environment_monitor_->UPDATE_ENVIRONMENT)` at the end of the function adding the box ?
If I do, then I get warnings of already existing links with the same name and although the scene is first displayed correctly then I think this is altering the program since the box won't attached to the eef and the robot completes his motion to the place pose without the box in his gripper.
If I don't, the behavior is as expected but the scene is first displayed when the robot starts his motion from the grasp pose to the place pose, which I find surprising...

IKapitaniuk, 2020-08-19:
Hi, 
I've encountered a similar problem with the warning: _Tried to add link (link1) with same name as an existing link._

I've looked a bit into the code and found the problem in the following function (tesseract_rosutils; utils.h; lines 1144-1152 ):
```
inline void toMsg(tesseract_msgs::TesseractState& state_msg, const tesseract_environment::Environment& env)
{
  state_msg.id = env.getName();
  state_msg.revision = static_cast<unsigned long>(env.getRevision());
  toMsg(state_msg.commands, env.getCommandHistory(), 0);

  tesseract_environment::EnvState::ConstPtr state = env.getCurrentState();
  toMsg(state_msg.joint_state, *state);
}
```

Particularly, the problem is in the line:
```
toMsg(state_msg.commands, env.getCommandHistory(), 0);
```
Are there any reasons why I should get **the full history of commands, starting from rev.0** in all successive messages?

In my case, I do two actions 1) Add a link, 2) Remove that link. The first step, and the corresponding state, is ok, but the second gives me a warning since the state message contains both commands to add and to remove the link. So, indeed, it tries to create the same link the second time.

Levi-Armstrong, 2020-08-19:
I believe the issues it that once the link has been added to the environment, the following commands should be moving the the link. Are you using the add command for attaching the box to the robot on pick? If so you need to use the move joint command which will just change the parent link of the joint. Now if you need to change the transform of the parent to origin you may need to leverage move link command where you provide a new joint for how it is attached to the gripper.

IKapitaniuk, 2020-08-19:
Ok, I've looked in details and found two related issues in two different places.
First of all, I've got that we, indeed, need to have the full history, to properly handle different revision numbers.

My setup:
1) There are 4 nodes: 
PathPlanningClient ->(Service ModifyEnvironment) -> tesseract_monitoring_environment_node(the standard one) -> (Topic TesseractState) PathPlanningServer + Rviz(in parallel with the Path PlanningServer) subscribed to the Tesseract State Topic

2) The client is just taken from _tesseract_ros/tesseract_ros/tesseract_monitoring/demos/demo_scene.cpp_ So, it just add the sphere to the environment and delete it after a while.

3) PathPlanningServer has the following callback:
```
  void envCallback(const tesseract_msgs::TesseractStatePtr msg)
  {
    tesseract_revision_ = msg->revision;
    
    const tesseract_environment::Environment::Ptr env = tesseract_->getEnvironment();
    if (!tesseract_rosutils::processMsg(env, *msg))
      ROS_ERROR_STREAM("Failed to update local Tesseract state");
  }
```

Every modification of the environment increases the revision number. I found that the logic of picking the corresponding commands to update _PathPlanningServer_ and Rviz contains some flaws.

The PathPlanningServer uses the following piece of code(tesseract_ros/tesseract_ros/tesseract_rosutils/include/tesseract_rosutils/utils.h, line 1380):
```
inline bool processMsg(tesseract_environment::Environment& env, const tesseract_msgs::TesseractState& state_msg)
{
  if (state_msg.id != env.getName() || static_cast<unsigned long>(env.getRevision()) > state_msg.revision)
    return false;

  // Only add new commands to environment
  if (env.getRevision() < static_cast<int>(state_msg.revision))
  {
    std::vector<tesseract_msgs::EnvironmentCommand> new_commands;
    new_commands.insert(new_commands.end(),
                        state_msg.commands.begin() + (env.getRevision() == 0 ? 0 : env.getRevision() - 1),
                        state_msg.commands.end());
    if (!processMsg(env, new_commands))
      return false;
  }
```
Here the insert command has the wrong limits. Revision number N contains N+1 commands, so if the previous revision was 3 and I want to process the revision 4, then I need to fetch the command with the index 4, but by following the current implementation it will be _env.getRevision() - 1 = 3_ and 4. Thus, I not only included the current command, but also one command before. That is the reason why I got the warning _Tried to add link (link1) with same name as an existing link._ when I was trying to delete the link1 after the creation on the previous step.

In my understanding, the condition should look like:
```
 new_commands.insert(new_commands.end(),
                        state_msg.commands.begin() + env.getRevision()),
                        state_msg.commands.end());
``` 

The similar logic for Rviz  widget(tesseract_ros/tesseract_ros/tesseract_rviz/src/render_tools/environment_widget.cpp, line 515):
```
void EnvironmentWidget::newTesseractStateCallback(const tesseract_msgs::TesseractStateConstPtr& state_msg)
{
  if (!tesseract_->getEnvironment())
    return;

  int current_version = tesseract_->getEnvironment()->getRevision();
  if (!tesseract_->getEnvironment() || state_msg->id != tesseract_->getEnvironment()->getName() ||
      current_version > static_cast<int>(state_msg->revision))
    return;

  if (update_state_)
    if (tesseract_rosutils::processMsg(tesseract_->getEnvironment(), state_msg->joint_state))
      update_required_ = true;

  if (static_cast<int>(state_msg->revision) > current_version)
  {
    std::vector<tesseract_msgs::EnvironmentCommand> commands;
    for (std::size_t i = (current_version == 0 ? 0 : static_cast<std::size_t>(current_version) - 1ul);
         i < state_msg->revision;
         ++i)
      commands.push_back(state_msg->commands[i]);

    if (!commands.empty())
      applyEnvironmentCommands(commands);
  }
...
}
```
I am not sure that here is the same revision number as for the _processMsg_ logic above, but the problem is the same. 
For every sequential revision, we get two commands rather than one. I can easily observe it in Rviz since the sphere is not deleted by the appropriate ModifyEnvironment message.

The following change works for me:
```
 for (std::size_t i =current_version;
         i < state_msg->revision;
         ++i)
```

What do you think in this regard?

Levi-Armstrong, 2020-08-19:
When the environment is create from a urdf it has zero commands. The commands are stored in  a std::vector which is zero indexed which is why the -1. The first command is added the revision would be 1 but the command would be stored in the vector add index 0.

Vector index | Revision
[0]                 |    1
[1]                 |    2
[2]                 |    3

IKapitaniuk, 2020-08-19:
Yes, regarding the indexes it is correct. But we go into these function when we got the message with the next revision, i.e. 
if we going to enter revision 2, our current revision is 1 and we need to take the [1] vector index to apply it.
So we don't need -1

Levi-Armstrong, 2020-08-20:
So, is there an issues on our end?

IKapitaniuk, 2020-08-20:
In my opinion, yes. 
Because now for each incremental revision update, two commands are fetched: the command from the previous revision and the actual one, though we need the second one only. 
It brings some problems: some modification commands can be executed twice, or, what I have observed, repeating the previous command can return an error/false state and the second, actual command, will be skipped until the next environment update. 
Since this logic is part of **tesseract_rosutils** and **tesseract_rviz** (I haven't looked for other places where the environment modification happens, so maybe some other packages are affected) it is an issue in tesseract_ros packages.

gavanderhoorn, 2020-08-20:
@IKapitaniuk: would it perhaps be possible to create a unit test which shows the problem?

That test would then succeed with your proposed patch, and fail with the current implementation.

Levi-Armstrong, 2020-08-20:
Thanks @gavanderhoorn. I think that would be good. We have been using this on multiple projects and I have not experience what you are explaining, but it may not be an obvious issue which I am missing.

IKapitaniuk, 2020-08-20:
@gavanderhoorn I think so, I will think how to make it. I tried to make it more obvious for this peace of code
```
for (std::size_t i = (current_version == 0 ? 0 : static_cast<std::size_t>(current_version) - 1ul);
         i < state_msg->revision;
         ++i)
```
Vector index | Revision
**[0] | 1** - we are here, i.e. **current version=1**
**[1] | 2** - new revision in message, i.e. **state_msg->revision = 2**
[2] | 3

To make it explicit:
```
for (i=current_version-1=0; i< 2; ++i)
```
So we take command [0] and [1], but we have to take only [1].

Levi-Armstrong, 2020-08-20:
Well not that I am looking at the code I believe you are correct. I am not sure how I have not ever ran into this.

Levi-Armstrong, 2020-08-20:
I think it would be best to move the logic to the ROS Utilities function and add a unit tests that validate this. Here is where this logic is currently used.

[image]

IKapitaniuk, 2020-08-20:
If you update your environment frequently enough, you won't see any problems, since, even in the worst case, the proper modification of the environment will happen at the following revision update.

Levi-Armstrong, 2020-08-20:
I have created a unit test that exposes the issue. I will push a patch soon.

Levi-Armstrong, 2020-08-20:
@IKapitaniuk Can you confirm that PR #34 solves the issue?