tesseract-robotics/tesseract_ros #17: Recommendation for faster cmake while developing examples

hzyjerry, 2020-05-18:
Hi, I was wondering if you have recommendations for making it faster to compile while building examples.

For instance, I am building examples in `tesseract_examples`, however every time I run
```bash
catkin build tesseract_ros_examples
```
It takes a couple of minutes, and packages such as `bullet3`, `tesseract_collision` are re-compiled as well.

It would be great to know if there is a way to skip rebuilding existing modules and incrementally re-compile. Thanks!

mpowelson, 2020-05-18:
Try `catkin build tesseract_ros_examples --no-deps `. That will build only that package. You can build it and its dependencies using `catkin build tesseract_ros_examples`

A more drastic solution is to put tesseract_ext in an extended workspace. Then things like Bullet which take a long time don't have to be rebuilt when you do a clean build of your development workspace.

hzyjerry, 2020-05-18:
This is great. Thanks so much!

closed 2020-05-18 by hzyjerry