tesseract-robotics/tesseract_ros #38: tesseract_ros examples produce strange output

ilya0ics, 2020-08-27:
Hi,

I have successfully compiled tesseract_ros with all dependencies, including tesseract_ext, on Ubuntu 18.04 + ROS Melodic. I have used catkin tools for compilation. 

I have tried to launch examples from tesseract_ros_examples package. However, all examples produce either no result at all or strange output

For example, basic_cartesian_example:

```
SUMMARY
========

`PARAMETERS`
 * /robot_description: <?xml version="1....
 * /robot_description_kinematics/manipulator/kinematics_solver: kdl_kinematics_pl...
 * /robot_description_kinematics/manipulator/kinematics_solver_attempts: 3
 * /robot_description_kinematics/manipulator/kinematics_solver_search_resolution: 0.005
 * /robot_description_kinematics/manipulator/kinematics_solver_timeout: 0.005
 * /robot_description_planning/joint_limits/joint_a1/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/joint_a1/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_a1/max_acceleration: 0
 * /robot_description_planning/joint_limits/joint_a1/max_velocity: 1.4834
 * /robot_description_planning/joint_limits/joint_a2/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/joint_a2/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_a2/max_acceleration: 0
 * /robot_description_planning/joint_limits/joint_a2/max_velocity: 1.4834
 * /robot_description_planning/joint_limits/joint_a3/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/joint_a3/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_a3/max_acceleration: 0
 * /robot_description_planning/joint_limits/joint_a3/max_velocity: 1.7452
 * /robot_description_planning/joint_limits/joint_a4/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/joint_a4/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_a4/max_acceleration: 0
 * /robot_description_planning/joint_limits/joint_a4/max_velocity: 1.3089
 * /robot_description_planning/joint_limits/joint_a5/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/joint_a5/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_a5/max_acceleration: 0
 * /robot_description_planning/joint_limits/joint_a5/max_velocity: 2.2688
 * /robot_description_planning/joint_limits/joint_a6/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/joint_a6/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_a6/max_acceleration: 0
 * /robot_description_planning/joint_limits/joint_a6/max_velocity: 2.356
 * /robot_description_planning/joint_limits/joint_a7/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/joint_a7/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_a7/max_acceleration: 0
 * /robot_description_planning/joint_limits/joint_a7/max_velocity: 2.356
 * /robot_description_semantic: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /tesseract_ros_examples_basic_cartesian_example_node/plotting: True
 * /tesseract_ros_examples_basic_cartesian_example_node/rviz: True
 * /trajopt_description: <...>

NODES
  /
    tesseract_ros_examples_basic_cartesian_example_node (tesseract_ros_examples/tesseract_ros_examples_basic_cartesian_example_node)
    tesseract_ros_examples_basic_cartesian_example_rviz (rviz/rviz)
    virtual_joint_broadcaster_0 (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [11035]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f1cf0bbe-e872-11ea-a3f6-6cb311254e73
process[rosout-1]: started with pid [11046]
started core service [/rosout]
process[virtual_joint_broadcaster_0-2]: started with pid [11053]
process[tesseract_ros_examples_basic_cartesian_example_node-3]: started with pid [11055]
process[tesseract_ros_examples_basic_cartesian_example_rviz-4]: started with pid [11060]
Warning: No version number was provided so latest parser will be used.
         at line 86 in /home/ilya/ros/workspaces/trajopt_ws/src/tesseract/tesseract/tesseract_scene_graph/src/srdf_model.cpp
[ WARN] [1598539098.599514132]: couldn't register subscriber on topic [/]
[ WARN] [1598539098.600756987]: couldn't register subscriber on topic [/]
[ WARN] [1598539098.610840608]: couldn't register subscriber on topic [/]
[ WARN] [1598539098.612104162]: couldn't register subscriber on topic [/]
[ WARN] [1598539098.863399895]: No version number was provided so latest parser will be used.
[ WARN] [1598539099.139625982]: No version number was provided so latest parser will be used.
[ERROR] [1598539099.401452199]: Hit enter key to step optimization!
```

The output is shown on the images below:

Initial state:

[image]

Final state:

[image]

It looks like the each robot state and tessaract trajectory is overlapped on previous ones in rviz making it difficult to understand the process. 
Additionally, /joint_states topic is empty.

### Same behavior is observed in pick_and_place_example:

```
SUMMARY
========

PARAMETERS
 * /box_parent_link: workcell_base
 * /box_side: 0.2
 * /box_x: 0.15
 * /box_y: 0.4
 * /robot_description: <?xml version="1....
 * /robot_description_semantic: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /tesseract_ros_examples_pick_and_place_example_node/file_write_cb: False
 * /tesseract_ros_examples_pick_and_place_example_node/plotting: True
 * /tesseract_ros_examples_pick_and_place_example_node/rviz: True
 * /tesseract_ros_examples_pick_and_place_example_node/steps_per_phase: 10

NODES
  /
    tesseract_ros_examples_pick_and_place_example_node (tesseract_ros_examples/tesseract_ros_examples_pick_and_place_example_node)
    tesseract_ros_examples_pick_and_place_example_rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [9627]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a6bca53a-e86f-11ea-a3f6-6cb311254e73
process[rosout-1]: started with pid [9638]
started core service [/rosout]
process[tesseract_ros_examples_pick_and_place_example_node-2]: started with pid [9645]
process[tesseract_ros_examples_pick_and_place_example_rviz-3]: started with pid [9646]
You can set logging level with TRAJOPT_LOG_THRESH. Valid values: FATAL ERROR WARN INFO DEBUG TRACE. Defaulting to ERROR
[ WARN] [1598537684.189004624]: couldn't register subscriber on topic [/]
[ WARN] [1598537684.190439744]: couldn't register subscriber on topic [/]
[ WARN] [1598537684.196596515]: couldn't register subscriber on topic [/]
[ WARN] [1598537684.197764790]: couldn't register subscriber on topic [/]
Warning: No version number was provided so latest parser will be used.
         at line 86 in /home/ilya/ros/workspaces/trajopt_ws/src/tesseract/tesseract/tesseract_scene_graph/src/srdf_model.cpp
[ INFO] [1598537684.320520394]: waitForService: Service [/get_tesseract_changes_rviz] has not been advertised, waiting...
[ WARN] [1598537684.988022010]: No version number was provided so latest parser will be used.
[ INFO] [1598537685.587379535]: waitForService: Service [/get_tesseract_changes_rviz] is now available.
[ WARN] [1598537686.231971919]: No version number was provided so latest parser will be used.
[ INFO] [1598537686.363397436]: Retrieve current environment changes!
[ INFO] [1598537706.469182163]: RViz environment Updated!
```
However, after going through optimization steps the node crash with following error:
```
tesseract_ros_examples_pick_and_place_example_node: /usr/include/eigen3/Eigen/src/Core/Block.h:147: Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::Block(XprType&, Eigen::Index, Eigen::Index, Eigen::Index, Eigen::Index) [with XprType = Eigen::Matrix<double, -1, -1, 1>; int BlockRows = -1; int BlockCols = -1; bool InnerPanel = false; Eigen::Index = long int]: Assertion `startRow >= 0 && blockRows >= 0 && startRow <= xpr.rows() - blockRows && startCol >= 0 && blockCols >= 0 && startCol <= xpr.cols() - blockCols' failed.
```
Initial state in rviz:

[image]

Final state:

[image]

Similar behavior is observed in all other tesseract_ros examples.

Levi-Armstrong, 2020-08-27:
In rviz you can change the trajectory visualization. The default shows every state of the trajectory, but you could set it show the trajectory motion in a loop or add  the trajectory player panel and you manually control what state of the trajectory you are viewing. Is this what you are looking for?

Related to the crashing we will need to look at why that is happening.

Levi-Armstrong, 2021-01-03:
This should be resolved in the feature/CommandLanguage branch which will be merged very soon. Please reopen if the issue exists with the latest code  in feature/CommandLanguage

closed 2021-01-03 by Levi-Armstrong