tesseract-robotics/tesseract_ros #39: Add getTesseractClone to environment monitor

mpowelson, 2020-09-03:
We could add a cache of Tesseract clones to the environment monitor. These could for example be stored in a map with the revision as the key. Whenever the environment changes, it would spin up a thread to refill the clone cache. Then when someone wants a Tesseract, all the getTesseractClone method has to do is check if the latest clone is done (if there is a key) and move it (and maybe set some quick last minute things like the current state). If there is no key we could just go ahead and clone it.

My thinking here is that global environment changes are probably happening out of sync with planning requests that might need to clone the Tesseract. It would be nice if the environment monitor kept a clone waiting so that it can grabbed quickly at planning time for instance.

mpowelson, 2020-09-03:
@Levi-Armstrong What do you think?

Levi-Armstrong, 2022-04-02:
This functionality is currently done within the planning server.

closed 2022-04-02 by Levi-Armstrong