tesseract-robotics/tesseract_ros #51: Motion Planning Issue within ROS

Levi-Armstrong, 2020-10-01:
Currently the position planners a Tesseract object which is not ideal because currently it could change during the motion planning because it is a pointer to the one stored in the monitor. Possible option to fix the issue. @mpowelson 

- Read Only Lock the environment once there is motion planning
- Clone the Tesseract
- Create data structure which contains clones of relavent data below
  - StateSolver
  - Contact Solvers
  -  etc.

mpowelson, 2020-10-02:
I think my preference would be to clone it. I think this goes hand in hand with the discussion we had somewhere about making the environment monitor keep a cache of clones ready to go, so you can get one quickly when a planning request comes in.

If we go that route, what if we made a generic CloneCache class? I imagine there are implementations out there if you know what to search for. Ignoring the details, I'm thinking something like below. The trick would be keeping the clones updated. Worst case we could just check if they are up to date and clone a new one if we need to, but I think we could do something more clever.

```
template <typename T>
class CloneCache {
public:
CloneCache()
{
  cloneInThread();
}

std::shared_ptr<T>& operator->() { return original; }

std::shared_ptr<T>  getClone()
{
  if(clones.back().needs_updating)
    update();
  return std::move(clones.back());
}
   
protected:
cloneInThread()  
{
clones.push_back(original->clone);
 }

std::shared_ptr<T> original_;
std::vector<std::shared_ptr<T>> clones;  // some container to hold them
};
```

Maybe this is more trouble than it is worth. I don't actually know how long it takes to clone the average Tesseract, but I feel that it could be significant - especially in the non-average case.

mpowelson, 2020-10-02:
It was #39

Levi-Armstrong, 2020-12-14:
This is addressed in PR #54

closed 2020-12-20 by Levi-Armstrong