tesseract-robotics/tesseract_ros #52: tesseract_monitoring doesn't export dependencies correctly

mpowelson, 2020-10-02:
It seems that when including tesseract_monitoring (specifically the environment monitor) in packages I also have to include the tesseract::tesseract_collision_core target as well as C++17 or else I get bullet errors. Is there anything we can do about that considering it is a catkin package?

Levi-Armstrong, 2020-10-02:
I am not sure, so the issue is that Bullet does not properly handle exporting its library for cmake when using find_package. May be able to switch to using the .pc which I think is correctly set.

Levi-Armstrong, 2020-10-02:
https://cmake.org/cmake/help/v3.0/module/FindPkgConfig.html

mpowelson, 2020-10-02:
I'm not sure I follow the problem. Targets that include tesseract::tesseract_collision_core seem ok, but our tesseract_monitoring libraries already include tesseract::tesseract which includes tesseract::tesseract_collision_core. Is it that double precision compile definition that is the problem?

Levi-Armstrong, 2020-10-02:
That is correct. Bullet also does not export some option libraries to its cmake config.

mpowelson, 2020-10-02:
Is there a way we can add that compile definition to our catkin libraries? The problem seems to be that if I use an environment monitor in a library, every library downstream from it then has to have tesseract::tesseract_collision_core manually added. 
`target_compile_definitions(${PROJECT_NAME}_environment PUBLIC "-DBT_USE_DOUBLE_PRECISION")` 

doesn't seem to do it and neither does adding

 `target_link_libraries(${PROJECT_NAME}_environment PUBLIC tesseract::tesseract tesseract::tesseract_collision_core ${catkin_LIBRARIES} ${Boost_LIBRARIES})`

mpowelson, 2020-10-02:
Actually, the root problem is `tesseract_ros/tesseract_rosutils/include/tesseract_rosutils/utils.h:84:0,`

Levi-Armstrong, 2020-10-02:
To add extra cmake you can provide CFG_EXTRAS in catkin_package.

mpowelson, 2020-10-02:
Ah ok. So I should add a file in tesseract_rosutils to set -DBT_USE_DOUBLE_PRECISION as a global CMake flag?

Levi-Armstrong, 2020-10-02:
Yea or we can fix it up stream. With switching to PkgConfig instead of find_package for bullet.

mpowelson, 2020-10-02:
Which would you prefer? I'd probably have to get you to do the upstream fix. Would that be replacing find_package(Bullet) with PkgConfig somehow in tesseract_collision?

Levi-Armstrong, 2020-10-02:
I think if you add the line below to tesseract_collision-config.cmake.in it may solve the issue. I will go through and switch it to using a target like what is done for octomap in the same file is the modern way to fix the issue.

```
include(${BULLET_USE_FILE})
```

Levi-Armstrong, 2020-10-02:
Need to add this to tesseract_collision-config.cmake

```
if(NOT TARGET tesseract_collision::bullet)
  add_library(tesseract_collision::bullet INTERFACE IMPORTED)
  set_target_properties(tesseract_collision::bullet PROPERTIES INTERFACE_INCLUDE_DIRECTORIES "${BULLET_INCLUDE_DIRS}")
  set_target_properties(tesseract_collision::bullet PROPERTIES INTERFACE_LINK_LIBRARIES "${BULLET_LIBRARIES}")
  set_target_properties(tesseract_collision::bullet PROPERTIES INTERFACE_COMPILE_DEFINITIONS "${BULLET_DEFINITIONS}")
endif()
```

Levi-Armstrong, 2020-10-02:
Testing things now.

mpowelson, 2020-10-02:
Ok. Thanks

Levi-Armstrong, 2020-10-02:
I just pushed a commit to this [PR](https://github.com/ros-industrial-consortium/tesseract/pull/399) which you should be able to cherry pick onto your branch and test out. It appears to work for me. Let me know if it solve the issue for you.

mpowelson, 2020-10-02:
That didn't seem to do it for me. I'm still getting `LinearMath/btConvexHullComputer.h: No such file or directory
 #include <LinearMath/btConvexHullComputer.h>`

Levi-Armstrong, 2020-10-03:
Are you using the latest tesseract_ros. The tesseract_rosutils should be linking against tesseract::tesseract_collision_core.

mpowelson, 2020-10-03:
I think I'm about a week behind the latest changes on Tesseract/Tesseract_ros. I'll update everything to the latest and try it again.

Levi-Armstrong, 2020-10-03:
Ok, I do remember having to add tesseract_collision to the tesseract_rosutils to solve some issue which may be what you are experiencing.

mpowelson, 2020-12-03:
Just reporting in that this is still an issue as of 80ba3c010d4454ed753a5d7b157fde170e3780d9 (11/21)

Levi-Armstrong, 2022-04-02:
I believe this has been resolved. Please reopen if the issue persists.

closed 2022-04-02 by Levi-Armstrong