tesseract-robotics/tesseract_ros #67: build/ runtime of tesseract_ros_examples package

ahedfares, 2020-12-08:
Hello,
I have recently cloned the default master branch package for tesseract_ros and installed all the dependencies with the proper mentioned branches in the dependencies.rosinstall file in order to explore the functionality developed for the online_planning example.
during build stage, i faced some errors that the compiler is unable to identify some Cmake macros (trajopt_target_compile_options) which i have solved by adding include(${CMAKE_INSTALL_PREFIX}/lib/cmake/trajopt_utils/trajopt_macros.cmake) in the cmakeLists.txt of the following packages:
trajopt, vhacd, trajopt_tools

the code is now building properly, but when i run any of the examples in the tesseract_ros_examples, i face the the following: 

[WARN] [1607435286.327775712]: couldn't register subscriber on topic [/]
[ERROR] [1607427947.791765422]: Error parsing urdf 'link' element for urdf 'online_planning_example'!
           Error parsing link 'visual' element for link 'human_estop_zone'!
             Error parsing visual 'geometry' element!
               Failed parsing geometry type cylinder!
                 Missing or failed parsing cylinder attribute radius!
[ERROR] [1607427947.791823172]: Failed to parse URDF.
[ERROR] [1607427947.792687676]: No environment
[ERROR] [1607427947.792774806]: No environment

i have traced the problem a bit and realised that the topics 
/tesseract/display_collisions and /tesseract_ros_examples_scene_graph_example_node/environment_monitor have no publishers, also the string passed to the function  EnvironmentWidget::onInitialize() is empty.

i would appreciate if you can direct me to the issue i am facing, i am not sure if it is related to my build or there are some modifications in the code that were not pushed to the repositories.

Thanking you in advance.

Levi-Armstrong, 2020-12-09:
Did you change the URDF for the example?

ahedfares, 2020-12-10:
no it is the exact same code as cloned from the repository.

Levi-Armstrong, 2020-12-13:
That is odd. Do you run into the same error with all of the examples?

ahedfares, 2020-12-18:
no, this problem happens for some of the examples where the xacro use a <geometry> attributes for sphere or cylinder that has a radius less than 1

i believe the issue is in the parser, if i replace the geometry shape or increase the radius to more than 1, it works.

Levi-Armstrong, 2020-12-18:
Thank you for the information. I will write a few unit tests for this case to see if I can reproduce.

Levi-Armstrong, 2020-12-30:
I will be adding unit tests to this [PR](https://github.com/ros-industrial-consortium/tesseract/pull/495) testing values less than 1 to see if there are issues.

Levi-Armstrong, 2020-12-30:
So I just ran the test with value less than one and they pass. It is very possible that a fix was applied to feature/CommandLanguage branch which will be merged into master soon.

closed 2021-02-02 by ahedfares