tesseract-robotics/tesseract_ros #79: Mimic joints result in "The complete state of the robot is not yet known" errors from environment monitor

mpowelson, 2021-01-21:


Levi-Armstrong, 2021-02-15:
Ah, the mimic joints have not been implemented.

mpowelson, 2021-02-18:
@Levi-Armstrong Would you mind commenting on what you think would be involved to fix this? Would this be a good first issue (e.g. for World ROS-I day 2021)?

Levi-Armstrong, 2021-02-20:
I was thinking about leveraging this library  to create a simple joint which wraps [this](https://github.com/codeplea/tinyexpr). I think we would just copy in the .h and .c into tesseract_urdf.

Levi-Armstrong, 2021-02-20:
Another option is [ExprTx](https://github.com/ArashPartow/exprtk) and we could just copy the .hpp file.

dave992, 2024-07-16:
Currently also running into this issue. Would be willing to contribute here if I can help.

I did find a lot of references to mimic joints in the code already, but I am not completely sure what changes are still needed. From what I observed: 
- [`isPassiveOrMimicDOF()`](https://github.com/tesseract-robotics/tesseract_ros/blob/master/tesseract_monitoring/src/current_state_monitor.cpp#L381) will need to be completed in order to prevent the warning
- The mimic joint needs to be changed to not be returned as a "active" joint, e.g. via `tesseract_environment::Environment::getActiveJointNames()`
- I assume the value for the mimic joint needs to be calculated and written to the environement somewhere in the environment monitor.

Levi-Armstrong, 2024-07-17:
I will need to spend some time looking into how best to support mimic joints

closed 2025-03-02 by Levi-Armstrong

rjoomen, 2025-03-03:
See #267 for `tesseract_ros`.