tesseract-robotics/tesseract_ros #94: planning failed

MinRen86, 2021-03-02:
I add a box at (-0.2, 0.4, 0.6) reference to "workcell_base" with size (0.5, 0.1, 0.1) in pick_and_place_example.

Everything is ok with the version "https://industrial-training-master.readthedocs.io/en/melodic/_source/demo3/Inspect-the-package.html".

Why planning is slow and will fail at place part with newest version?

Levi-Armstrong, 2021-03-02:
I assume you are changing the parameter by passing the information to the roslaunch file. Can you provide the command line instruction you executed?

MinRen86, 2021-03-03:
I did not change the launch file of the example.
The following is my executing command line message:

... logging to /home/minren/.ros/log/f4a8b1d4-7c04-11eb-b575-6045cbc48079/roslaunch-minren-21774.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://minren:45763/

SUMMARY
========

PARAMETERS
 * /box_side: 0.2
 * /box_x: 0.15
 * /box_y: 0.4
 * /robot_description: <?xml version="1....
 * /robot_description_semantic: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /tesseract_ros_examples_pick_and_place_example_node/file_write_cb: False
 * /tesseract_ros_examples_pick_and_place_example_node/plotting: True
 * /tesseract_ros_examples_pick_and_place_example_node/rviz: True
 * /tesseract_ros_examples_pick_and_place_example_node/steps_per_phase: 10

NODES
  /
    tesseract_ros_examples_pick_and_place_example_node (tesseract_ros_examples/tesseract_ros_examples_pick_and_place_example_node)
    tesseract_ros_examples_pick_and_place_example_rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [21791]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f4a8b1d4-7c04-11eb-b575-6045cbc48079
process[rosout-1]: started with pid [21804]
started core service [/rosout]
process[tesseract_ros_examples_pick_and_place_example_node-2]: started with pid [21813]
process[tesseract_ros_examples_pick_and_place_example_rviz-3]: started with pid [21814]
You can set logging level with TRAJOPT_LOG_THRESH. Valid values: FATAL ERROR WARN INFO DEBUG TRACE. Defaulting to ERROR
Warning: SRDF Parser: No version number was provided so latest parser (version: 1.0.0) will be used.
         at line 92 in /home/minren/new_trajopt/src/tesseract/tesseract_scene_graph/src/srdf_model.cpp
Debug:   SRDF Parser: The version number warning can be suppressed by adding the attribute: version=1.0.0
Debug:   Created collision object for link iiwa_link_4
Debug:   Created collision object for link iiwa_link_0
Debug:   Created collision object for link iiwa_link_5
Debug:   Created collision object for link iiwa_link_1
Debug:   Created collision object for link iiwa_link_2
Debug:   Created collision object for link workcell_base
Debug:   Created collision object for link iiwa_link_7
Debug:   Created collision object for link iiwa_link_3
Debug:   Created collision object for link iiwa_link_6
Debug:   Created collision object for link cabinet_1
Debug:   Created collision object for link iiwa_link_4
Debug:   Created collision object for link iiwa_link_0
Debug:   Created collision object for link iiwa_link_5
Debug:   Created collision object for link iiwa_link_1
Debug:   Created collision object for link iiwa_link_2
Debug:   Created collision object for link workcell_base
Debug:   Created collision object for link iiwa_link_7
Debug:   Created collision object for link iiwa_link_3
Debug:   Created collision object for link iiwa_link_6
Debug:   Created collision object for link cabinet_1
[ INFO] [1614764635.554581629]: Discrete Contact Monitor Registered: tesseract_collision::tesseract_collision_bullet::BulletDiscreteBVHManager
[ INFO] [1614764635.555860382]: Discrete Contact Monitor Registered: tesseract_collision::tesseract_collision_bullet::BulletDiscreteSimpleManager
[ INFO] [1614764635.555913748]: Discrete Contact Monitor Registered: tesseract_collision::tesseract_collision_fcl::FCLDiscreteBVHManager
[ INFO] [1614764635.556868121]: Continuous Contact Monitor Registered: tesseract_collision::tesseract_collision_bullet::BulletCastBVHManager
[ INFO] [1614764635.556951991]: Continuous Contact Monitor Registered: tesseract_collision::tesseract_collision_bullet::BulletCastSimpleManager
[... 2141 lines cut ...]
   collision_13 | ---------- |  0.000e+00 |  0.000e+00 |  4.657e-09 |  0.000e+00 |   ------  
   collision_14 | ---------- |  0.000e+00 |  0.000e+00 | -3.004e-07 |  0.000e+00 | -0.000e+00
   collision_15 | ---------- |  0.000e+00 |  0.000e+00 | -1.952e-05 |  0.000e+00 | -0.000e+00
   collision_16 | ---------- |  0.000e+00 |  0.000e+00 |  1.510e-05 |  0.000e+00 |  0.000e+00
   collision_17 | ---------- |  0.000e+00 |  2.497e-05 |  1.047e-08 | -2.497e-05 | -2.384e+03
   collision_18 | ---------- |  0.000e+00 |  2.497e-05 | -1.362e-06 | -2.497e-05 |  1.833e+01
   collision_19 | ---------- |  0.000e+00 |  0.000e+00 | -1.002e-06 |  0.000e+00 | -0.000e+00
   collision_20 | ---------- |  0.000e+00 |  0.000e+00 |  1.271e-08 |  0.000e+00 |  0.000e+00
   collision_21 | ---------- |  0.000e+00 |  0.000e+00 |  5.792e-07 |  0.000e+00 |  0.000e+00
   collision_22 | ---------- |  0.000e+00 |  0.000e+00 |  9.494e-07 |  0.000e+00 |  0.000e+00
   collision_23 | ---------- |  0.000e+00 |  0.000e+00 |  1.012e-06 |  0.000e+00 |  0.000e+00
   collision_24 | ---------- |  0.000e+00 |  0.000e+00 | -2.097e-06 |  0.000e+00 | -0.000e+00
   collision_25 | ---------- |  0.000e+00 |  0.000e+00 | -6.916e-06 |  0.000e+00 | -0.000e+00
   collision_26 | ---------- |  0.000e+00 |  0.000e+00 | -8.645e-06 |  0.000e+00 | -0.000e+00
   collision_27 | ---------- |  0.000e+00 |  0.000e+00 | -5.870e-06 |  0.000e+00 | -0.000e+00
   collision_28 | ---------- |  0.000e+00 |  0.000e+00 |  8.515e-08 |  0.000e+00 |  0.000e+00
   collision_29 | ---------- |  1.425e-05 |  1.538e-06 |  1.420e-05 |  1.271e-05 |  8.953e-01
[... cut ...]
   collision_32 | ---------- |  3.068e+00 |  3.107e+00 | -3.893e-02 | -3.951e-02 |  1.015e+00
   collision_33 | ---------- |  5.163e+00 |  5.157e+00 |  6.842e-03 |  6.397e-03 |  9.350e-01
   collision_34 | ---------- |  7.096e+00 |  7.114e+00 | -1.363e-02 | -1.764e-02 |  1.294e+00
   collision_35 | ---------- |  1.370e+01 |  1.372e+01 | -2.679e-02 | -2.660e-02 |  9.929e-01
   collision_36 | ---------- |  1.629e+01 |  1.629e+01 |  4.313e-03 |  4.946e-03 |  1.147e+00
   collision_37 | ---------- |  1.755e+01 |  2.341e+01 |  6.215e-03 | -5.861

MinRen86, 2021-03-03:
I did not change the launch file of the example.
The following is my executing command line message:

... logging to /home/minren/.ros/log/f4a8b1d4-7c04-11eb-b575-6045cbc48079/roslaunch-minren-21774.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://minren:45763/

PARAMETERS
 * /box_side: 0.2
 * /box_x: 0.15
 * /box_y: 0.4
 * /robot_description: <?xml version="1....
 * /robot_description_semantic: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /tesseract_ros_examples_pick_and_place_example_node/file_write_cb: False
 * /tesseract_ros_examples_pick_and_place_example_node/plotting: True
 * /tesseract_ros_examples_pick_and_place_example_node/rviz: True
 * /tesseract_ros_examples_pick_and_place_example_node/steps_per_phase: 10

NODES
  /
    tesseract_ros_examples_pick_and_place_example_node (tesseract_ros_examples/tesseract_ros_examples_pick_and_place_example_node)
    tesseract_ros_examples_pick_and_place_example_rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [21791]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f4a8b1d4-7c04-11eb-b575-6045cbc48079
process[rosout-1]: started with pid [21804]
started core service [/rosout]
process[tesseract_ros_examples_pick_and_place_example_node-2]: started with pid [21813]
process[tesseract_ros_examples_pick_and_place_example_rviz-3]: started with pid [21814]
You can set logging level with TRAJOPT_LOG_THRESH. Valid values: FATAL ERROR WARN INFO DEBUG TRACE. Defaulting to ERROR
Warning: SRDF Parser: No version number was provided so latest parser (version: 1.0.0) will be used.
         at line 92 in /home/minren/new_trajopt/src/tesseract/tesseract_scene_graph/src/srdf_model.cpp
Debug:   SRDF Parser: The version number warning can be suppressed by adding the attribute: version=1.0.0
Debug:   Created collision object for link iiwa_link_4
Debug:   Created collision object for link iiwa_link_0
Debug:   Created collision object for link iiwa_link_5
Debug:   Created collision object for link iiwa_link_1
Debug:   Created collision object for link iiwa_link_2
Debug:   Created collision object for link workcell_base
Debug:   Created collision object for link iiwa_link_7
Debug:   Created collision object for link iiwa_link_3
Debug:   Created collision object for link iiwa_link_6
Debug:   Created collision object for link cabinet_1
Debug:   Created collision object for link iiwa_link_4
Debug:   Created collision object for link iiwa_link_0
Debug:   Created collision object for link iiwa_link_5
Debug:   Created collision object for link iiwa_link_1
Debug:   Created collision object for link iiwa_link_2
Debug:   Created collision object for link workcell_base
Debug:   Created collision object for link iiwa_link_7
Debug:   Created collision object for link iiwa_link_3
Debug:   Created collision object for link iiwa_link_6
Debug:   Created collision object for link cabinet_1
[ INFO] [1614764635.554581629]: Discrete Contact Monitor Registered: tesseract_collision::tesseract_collision_bullet::BulletDiscreteBVHManager
[ INFO] [1614764635.555860382]: Discrete Contact Monitor Registered: tesseract_collision::tesseract_collision_bullet::BulletDiscreteSimpleManager
[ INFO] [1614764635.555913748]: Discrete Contact Monitor Registered: tesseract_collision::tesseract_collision_fcl::FCLDiscreteBVHManager
[ INFO] [1614764635.556868121]: Continuous Contact Monitor Registered: tesseract_collision::tesseract_collision_bullet::BulletCastBVHManager
[ INFO] [1614764635.556951991]: Continuous Contact Monitor Registered: tesseract_collision::tesseract_collision_bullet::BulletCastSimpleManager
[ INFO] [1614764635.561208232]: Publishing maintained environment on '/tesseract_ros_examples/tesseract_published_environment'
Debug:   Created collision object for link box
Debug:   Created collision object for link box
[... 2138 lines cut ...]
   collision_13 | ---------- |  0.000e+00 |  0.000e+00 |  4.657e-09 |  0.000e+00 |   ------  
   collision_14 | ---------- |  0.000e+00 |  0.000e+00 | -3.004e-07 |  0.000e+00 | -0.000e+00
   collision_15 | ---------- |  0.000e+00 |  0.000e+00 | -1.952e-05 |  0.000e+00 | -0.000e+00
   collision_16 | ---------- |  0.000e+00 |  0.000e+00 |  1.510e-05 |  0.000e+00 |  0.000e+00
   collision_17 | ---------- |  0.000e+00 |  2.497e-05 |  1.047e-08 | -2.497e-05 | -2.384e+03
   collision_18 | ---------- |  0.000e+00 |  2.497e-05 | -1.362e-06 | -2.497e-05 |  1.833e+01
   collision_19 | ---------- |  0.000e+00 |  0.000e+00 | -1.002e-06 |  0.000e+00 | -0.000e+00
   collision_20 | ---------- |  0.000e+00 |  0.000e+00 |  1.271e-08 |  0.000e+00 |  0.000e+00
   collision_21 | ---------- |  0.000e+00 |  0.000e+00 |  5.792e-07 |  0.000e+00 |  0.000e+00
   collision_22 | ---------- |  0.000e+00 |  0.000e+00 |  9.494e-07 |  0.000e+00 |  0.000e+00
   collision_23 | ---------- |  0.000e+00 |  0.000e+00 |  1.012e-06 |  0.000e+00 |  0.000e+00
   collision_24 | ---------- |  0.000e+00 |  0.000e+00 | -2.097e-06 |  0.000e+00 | -0.000e+00
   collision_25 | ---------- |  0.000e+00 |  0.000e+00 | -6.916e-06 |  0.000e+00 | -0.000e+00
   collision_26 | ---------- |  0.000e+00 |  0.000e+00 | -8.645e-06 |  0.000e+00 | -0.000e+00
   collision_27 | ---------- |  0.000e+00 |  0.000e+00 | -5.870e-06 |  0.000e+00 | -0.000e+00
[... cut ...]
   collision_32 | ---------- |  3.068e+00 |  3.107e+00 | -3.893e-02 | -3.951e-02 |  1.015e+00
   collision_33 | ---------- |  5.163e+00 |  5.157e+00 |  6.842e-03 |  6.397e-03 |  9.350e-01
   collision_34 | ---------- |  7.096e+00 |  7.114e+00 | -1.363e-02 | -1.764e-02 |  1.294e+00
   collision_35 | ---------- |  1.370e+01 |  1.372e+01 | -2.679e-02 | -2.660e-02 |  9.929e-01
   collision_36 | ---------- |  1.629e+01 |  1.629e+01 |  4.313e-03 |  4.946e-03 |  1.147e+00
   collision_37 | ---------- |  1.755e+01 |  2.341e+01 |  6.215e-03 | -5.861e+00 | -9.431e+02

MinRen86, 2021-03-03:
I didn't change anything in launch file. I just use the command: `roslaunch tesseract_ros_examples pick_and_place_example.launch`
But I add some code in *pick_and_place_example.cpp*.
In the newest version, I add a box as an obstacle by the following code:

    {  
      Link link_box("box1");

      Visual::Ptr visual = std::make_shared<Visual>();
      visual->origin = Eigen::Isometry3d::Identity();
      visual->geometry = std::make_shared<tesseract_geometry::Box>(0.5, 0.1, 0.1);
      link_box.visual.push_back(visual);

      Collision::Ptr collision = std::make_shared<Collision>();
      collision->origin = visual->origin;
      collision->geometry = visual->geometry;
      link_box.collision.push_back(collision);

      Joint joint_box("joint_box1");
      joint_box.parent_link_name = "workcell_base";
      joint_box.child_link_name = "box1";
      joint_box.type = JointType::FIXED;
      joint_box.parent_to_joint_origin_transform = Eigen::Isometry3d::Identity();
      joint_box.parent_to_joint_origin_transform.translation() += Eigen::Vector3d(-0.2, 0.4, 0.6);

      auto cmd = std::make_shared<tesseract_environment::AddLinkCommand>(link_box, joint_box);

      if (!monitor_->applyCommand(cmd))
        return false;
    }
Planning time is much longer and planning will fail at place part.
However, everything is fine when I do the same thing in old version I mentioned before.

Levi-Armstrong, 2021-03-03:
I will take a look and see what is happening.

Levi-Armstrong, 2021-03-03:
Do you have a video or image of the result trajectory for the old version? I would interested to see what is found for a solution.

MinRen86, 2021-03-04:
video:

https://user-images.githubusercontent.com/74589814/109913072-406bde00-7ce8-11eb-96a8-d8ce390a8122.mp4

path of PLACE:

 -0.275713   0.670101 -0.0664011   -1.90516  0.0765914   0.568733    1.17844          2
 -0.289731   0.629265  -0.114438   -1.90253  0.0928637   0.608498    1.16489    9.04852
 -0.254122    0.59585  -0.163524   -1.90012   0.109136   0.648263    1.15133    9.04852
 -0.218512   0.562436   -0.21261   -1.89771   0.125408   0.688028    1.13777    9.04852
 -0.182902   0.529024   -0.26169   -1.89529    0.14168   0.727793    1.12422    9.04852
 -0.147267   0.495655  -0.310731   -1.89283   0.157953   0.767558    1.11066    9.04852
 -0.106425   0.466628  -0.355322   -1.88655   0.174225   0.807323    1.09711    9.04852
-0.0627547   0.439109  -0.398066   -1.87929   0.190497   0.847088    1.08355    9.04852
-0.0182585   0.412169  -0.440244   -1.87169   0.206768   0.886853       1.07    9.04852
 0.0460062   0.411012  -0.458579    -1.8545   0.223038   0.926617    1.05644    9.04852
  0.106514   0.407775  -0.570702   -1.90403   0.241037    0.79933     1.1607    9.04852
  0.182346   0.429967  -0.657439   -1.94115   0.259025   0.672043    1.26495    9.04852
    0.2589   0.453051  -0.743534   -1.97768      0.277   0.544756     1.3692    9.04852
  0.337532   0.479326  -0.827855   -2.01222   0.294975    0.41747    1.47345    9.04852
  0.419154   0.508322  -0.909393   -2.04293   0.312482   0.290186    1.57771    9.04852
  0.503477    0.54329  -0.987913   -2.06833   0.329961   0.162903    1.68196    9.04852
  0.587806   0.578258   -1.06643   -2.09374   0.347467  0.0356226     1.7862    9.04852
  0.681987   0.659533   -1.12678    -2.0944   0.365895 -0.0916536    1.89045    9.04852
   0.76772   0.764442   -1.17225    -2.0944   0.374359  -0.218922    1.99468    9.04852
  0.893096   0.865258   -1.21179   -2.09438   0.364827  -0.346169    2.09888    9.04852
   1.00185   0.899172   -1.27711   -2.01636   0.353442  -0.473696    2.20434    9.04852
   1.12027   0.987173   -1.16384   -1.81493   0.232061  -0.632565    2.26681    9.01542
   1.08184    1.04413   -1.18401   -1.73822   0.204477  -0.598251     2.3383    9.04852
    1.0404    1.09529   -1.20074   -1.65858   0.180702  -0.563995    2.40163    9.04852
  0.997895    1.14533   -1.21952   -1.57558    0.15753  -0.528365    2.46566    9.04852
   0.95363    1.19374   -1.23993    -1.4887   0.134871  -0.491121    2.52989    9.04852
  0.906997    1.24021   -1.26181   -1.39723   0.112467  -0.451884    2.59424    9.04852
  0.855991    1.28014   -1.27871    -1.3002  0.0942906  -0.410554    2.64763    9.04852
  0.799333    1.31078   -1.28473   -1.19629  0.0847774  -0.366387     2.6794    9.04852
  0.737495     1.3366   -1.28372   -1.08352  0.0822511  -0.317794    2.69587    9.04852

Levi-Armstrong, 2021-03-04:
Thank you for the video it is very helpful. There are two difference between the old version and the current. The first difference is this old is using BMPD solver and the new version is using OSQP due to license. The second difference is the old version by default would seed the trajectory with the start state. The new version allows you to define a simple planner which controls this. I will first try switching the to BMPD and if that does not change the results I will change the initial trajectory seed to use the start state.

Levi-Armstrong, 2022-04-02:
The example have been updated and all run so closing. Please reopen if you are still having issues.

closed 2022-04-02 by Levi-Armstrong

MinRen86, 2022-04-03:
Dear Armstrong: How do you fix this issue?

Levi-Armstrong, 2022-04-05:
This fixes the existing example but did not test your specific example with the box added.