tesseract-robotics/tesseract_ros #102: Building the repo, cannot find `BulletInverseDynamics-float64`

mohamedsayed18, 2021-04-04:
I'm trying to build the repo, I followed the following steps
1. `mkdir -p tesseract_planner/src`
2. cd tesseract_planner/src
3. `git clone --depth=1 git@github.com:ros-industrial-consortium/tesseract.git`
4. `cd ../..`
5. `catkin build -DCMAKE_BUILD_TYPE=Release -DTESSERACT_PACKAGE=ON`
```
Starting  >>> tesseract_collision                                              
_______________________________________________________________________________
Errors     << tesseract_collision:make /media/mohamed/C03CCDB43CCDA62E/work/tessract_space/logs/tesseract_collision/build.make.001.log
/usr/bin/ld: cannot find -lBulletInverseDynamics-float64
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/create_convex_hull.dir/build.make:94: create_convex_hull] Error 1
make[1]: *** [CMakeFiles/Makefile2:270: CMakeFiles/create_convex_hull.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: cannot find -lBulletInverseDynamics-float64
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/tesseract_collision_fcl.dir/build.make:126: libtesseract_collision_fcl.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:324: CMakeFiles/tesseract_collision_fcl.dir/all] Error 2
/usr/bin/ld: cannot find -lBulletInverseDynamics-float64
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/tesseract_collision_bullet.dir/build.make:228: libtesseract_collision_bullet.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:297: CMakeFiles/tesseract_collision_bullet.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
cd /media/mohamed/C03CCDB43CCDA62E/work/tessract_space/build/tesseract_collision; catkin build --get-env tesseract_collision | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
...............................................................................
Failed     << tesseract_collision:make                        [ Exited with code 2 ]
Failed    <<< tesseract_collision                             [ 0.8 seconds ]  
Abandoned <<< tesseract_urdf                                  [ Unrelated job failed ]
Abandoned <<< tesseract_kinematics                            [ Unrelated job failed ]
Abandoned <<< tesseract_environment                           [ Unrelated job failed ]
Abandoned <<< tesseract_visualization                         [ Unrelated job failed ]
[build] Summary: 7 of 12 packages succeeded.                                   
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 4 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 11.2 seconds total.
```
I tried to install the build the bullet [library](https://github.com/bulletphysics/bullet3) I install it using vcpkg but I'm still facing same error, am I using the right bullet library or `BulletInverseDynamics-float64` is from a different repo.

Notes:
I have the fcl and ros_industrial_cmake_boilerplate as they are specified in the Cmake
OS: Ubuntu 20.04.2 LTS

Levi-Armstrong, 2021-04-05:
Did you install Bullet using the command below?

``` bash
 sudo apt install libbullet-dev
```

mohamedsayed18, 2021-04-05:
> Did you install Bullet using the command below?
> 
> ```shell
>  sudo apt install libbullet-dev
> ```

Yes, and this is the output `libbullet-dev is already the newest version (2.87+dfsg-2).`
I tried to rebuild but I have the same error.

Levi-Armstrong, 2021-04-05:
Oh, did you also install `sudo apt install libbullet-extras-dev`.

Levi-Armstrong, 2021-04-05:
Does the following file exist on your machine? `/usr/lib/x86_64-linux-gnu/libBulletDynamics-float64.so`

mohamedsayed18, 2021-04-05:
> Oh, did you also install `sudo apt install libbullet-extras-dev`.

I think that what I was missing. After installing it the bulletinverse error disapper but I still got this error
```
Starting  >>> tesseract_collision                                              
_______________________________________________________________________________
Errors     << tesseract_collision:make /media/mohamed/C03CCDB43CCDA62E/work/tessract_space/logs/tesseract_collision/build.make.001.log
/media/mohamed/C03CCDB43CCDA62E/work/tessract_space/src/tesseract/tesseract_collision/src/bullet/tesseract_convex_convex_algorithm.cpp: In member function ‘virtual void tesseract_collision::tesseract_collision_bullet::TesseractConvexConvexAlgorithm::processCollision(const btCollisionObjectWrapper*, const btCollisionObjectWrapper*, const btDispatcherInfo&, btManifoldResult*)’:
/media/mohamed/C03CCDB43CCDA62E/work/tessract_space/src/tesseract/tesseract_collision/src/bullet/tesseract_convex_convex_algorithm.cpp:689:26: error: ‘class btConvexPolyhedron’ has no member named ‘initialize2’; did you mean ‘initialize’?
               polyhedron.initialize2();
                          ^~~~~~~~~~~
                          initialize
/media/mohamed/C03CCDB43CCDA62E/work/tessract_space/src/tesseract/tesseract_collision/src/bullet/tesseract_convex_convex_algorithm.cpp:691:28: error: ‘class btPolyhedralConvexShape’ has no member named ‘setPolyhedralFeatures’; did you mean ‘initializePolyhedralFeatures’?
               polyhedronB->setPolyhedralFeatures(polyhedron);
                            ^~~~~~~~~~~~~~~~~~~~~
                            initializePolyhedralFeatures
make[2]: *** [CMakeFiles/tesseract_collision_bullet.dir/src/bullet/tesseract_convex_convex_algorithm.cpp.o] Error 1
make[1]: *** [CMakeFiles/tesseract_collision_bullet.dir/all] Error 2
make: *** [all] Error 2
cd /media/mohamed/C03CCDB43CCDA62E/work/tessract_space/build/tesseract_collision; catkin build --get-env tesseract_collision | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << tesseract_collision:make                        [ Exited with code 2 ]
Failed    <<< tesseract_collision                             [ 14.9 seconds ] 
Abandoned <<< tesseract_urdf                                  [ Unrelated job failed ]
Abandoned <<< tesseract_kinematics                            [ Unrelated job failed ]
Abandoned <<< tesseract_environment                           [ Unrelated job failed ]
Abandoned <<< tesseract_visualization                         [ Unrelated job failed ]
[build] Summary: 7 of 12 packages succeeded.                                   
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 4 packages were abandoned.                                
[build]   Failed:    1 packages failed.
```
should I edit it as suggested?

mohamedsayed18, 2021-04-05:
> Does the following file exist on your machine? `/usr/lib/x86_64-linux-gnu/libBulletDynamics-float64.so`

The file I have is `libBulletDynamics.so`

Levi-Armstrong, 2021-04-06:
Did you install it from source? If so, what branch did you use? Everything should build using the system version on 20.04 so if you did install from source you may uninstall it.

mohamedsayed18, 2021-04-06:
> Did you install it from source? If so, what branch did you use? Everything should build using the system version on 20.04 so if you did install from source you may uninstall it.

The last error was on ubuntu 16 version.
On the 20.04 version, I recloned the repos from source tesseract_ros, tesseract(master branch)
and build them using this command `catkin build -DTESSERACT_ENABLE_CLANG_TIDY=ON`
```
Errors     << tesseract_urdf:make /media/mohamed/C03CCDB43CCDA62E/work/tessract20/logs/tesseract_urdf/build.make.000.log                     
In file included from /media/mohamed/C03CCDB43CCDA62E/tutorials/Innopolis/Year_2/Project/drone_ws/devel/include/eigen3/Eigen/StdVector:23,
                 from /usr/include/pcl-1.10/pcl/point_cloud.h:45,
                 from /usr/include/pcl-1.10/pcl/io/pcd_io.h:43,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract/tesseract_urdf/include/tesseract_urdf/point_cloud.h:35,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract/tesseract_urdf/include/tesseract_urdf/octomap.h:42,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract/tesseract_urdf/include/tesseract_urdf/geometry.h:48,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract/tesseract_urdf/include/tesseract_urdf/visual.h:42,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract/tesseract_urdf/include/tesseract_urdf/link.h:40,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract/tesseract_urdf/include/tesseract_urdf/urdf_parser.h:41,
                 from /media/mohamed/C03CCDB43CCDA62E/work/tessract20/src/tesseract/tesseract_urdf/src/urdf_parser.cpp:27:
/media/mohamed/C03CCDB43CCDA62E/tutorials/Innopolis/Year_2/Project/drone_ws/devel/include/eigen3/Eigen/src/StlSupport/StdVector.h:69:9: error: partial specialization of ‘class std::vector<T, Eigen::aligned_allocator<T> >’ after instantiation of ‘class std::vector<tesseract_collision::ContactResult, Eigen::aligned_allocator<tesseract_collision::ContactResult> >’ [-fpermissive]
   class vector<T,EIGEN_ALIGNED_ALLOCATOR<T> >
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/mohamed/C03CCDB43CCDA62E/tutorials/Innopolis/Year_2/Project/drone_ws/devel/include/eigen3/Eigen/src/StlSupport/StdVector.h:69:9: error: partial specialization of ‘class std::vector<T, Eigen::aligned_allocator<T> >’ after instantiation of ‘class std::vector<Eigen::Matrix<double, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 2, 1> > >’ [-fpermissive]
/media/mohamed/C03CCDB43CCDA62E/tutorials/Innopolis/Year_2/Project/drone_ws/devel/include/eigen3/Eigen/src/StlSupport/StdVector.h:69:9: error: partial specialization of ‘class std::vector<T, Eigen::aligned_allocator<T> >’ after instantiation of ‘class std::vector<Eigen::Matrix<double, 4, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 4, 1> > >’ [-fpermissive]
make[2]: *** [CMakeFiles/tesseract_urdf.dir/build.make:63: CMakeFiles/tesseract_urdf.dir/src/urdf_parser.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:101: CMakeFiles/tesseract_urdf.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
cd /media/mohamed/C03CCDB43CCDA62E/work/tessract20/build/tesseract_urdf; catkin build --get-env tesseract_urdf | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
.............................................................................................................................................
Failed     << tesseract_urdf:make                             [ Exited with code 2 ]                                                         
Failed    <<< tesseract_urdf                                  [ 20.0 seconds ]                                                               
Abandoned <<< tesseract_kinematics                            [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_environment                           [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_visualization                         [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_rosutils                              [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_monitoring                            [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_planning_server                       [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_ros_examples                          [ Unrelated job failed ]                                                       
Abandoned <<< tesseract_rviz                                  [ Unrelated job failed ]                                                       
[build] Summary: 9 of 18 packages succeeded.                                                                                                 
[build]   Ignored:   None.                                                                                                                   
[build]   Warnings:  1 packages succeeded with warnings.                                                                                     
[build]   Abandoned: 8 packages were abandoned.                                                                                              
[build]   Failed:    1 packages failed.                                                                                                      
[build] Runtime: 37.4 seconds total.                                                                                                         
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```
I installed libbullet-dev, libbullet-extras-dev ros-noetic-fcl taskflow, am I missing another package?

Levi-Armstrong, 2021-04-06:
Ah, this error is because you have enabled clang tidy and not building in debug. I would do a clean and then`catkin build`.

mohamedsayed18, 2021-04-07:
I did the following 
`catkin clean`
`catkin build`
but I still have the same error as above

Levi-Armstrong, 2021-04-07:
Are you using the system version of Eigen?

mohamedsayed18, 2021-04-07:
> Are you using the system version of Eigen?

I checked my version using `pkg-config --modversion eigen3`
output: 3.2.7

Levi-Armstrong, 2021-04-08:
I recommend trying the system version if you can which is 3.3.7.

mohamedsayed18, 2021-04-09:
I installed the [Eigen 3.3.7](https://gitlab.com/libeigen/eigen/-/releases/3.3.7), extract it, and build it using CMake. but I still have the same error.
I think there is a conflict in Eigen version but I don't know how to solve
when I try this command  `grep "#define EIGEN_[^_]*_VERSION" /usr/local/include/eigen3/Eigen/src/Core/util/Macros.h`
```
#define EIGEN_WORLD_VERSION 3
#define EIGEN_MAJOR_VERSION 3
#define EIGEN_MINOR_VERSION 7
```
but when I try this `pkg-config --modversion eigen3`
output: `3.2.7`

mohamedsayed18, 2021-04-09:
I think [this](https://stackoverflow.com/questions/34256027/eigen-allocator-causing-trouble-in-templated-class) is a similar problem to what I have but I don't know how to apply the solution.
Also [here](https://stackoverflow.com/questions/34256027/eigen-allocator-causing-trouble-in-templated-class) I tried the suggested solution but it didn't work.
One solution I found on the internet is to comment the `#include <Eigen/StdVector>`. it worked for me but I'm still working on fixing some build errors related to Trajopt

mohamedsayed18, 2021-04-09:
If I kept the `#include <Eigen/StdVector>` and I set the Eigen variable globally but still the same error
`export EIGEN3_INCLUDE_DIR=/media/mohamed/C03CCDB43CCDA62E/work/Code/eigen-3.3.7/Eigen/`

mohamedsayed18, 2021-04-10:
It was a problem with Eigen library
```
export EIGEN3_INCLUDE_DIR=/media/mohamed/C03CCDB43CCDA62E/work/Code/eigen-3.3.7/
export EIGEN_INCLUDE_DIR=/media/mohamed/C03CCDB43CCDA62E/work/Code/eigen-3.3.7/
```
I set these environment variables in my bashrc and it solved the above issue but I met another [error](https://github.com/ros-industrial-consortium/tesseract_planning/issues/70) with tesseract_planning

closed 2021-04-12 by mohamedsayed18