tesseract-robotics/tesseract_ros #105: Run online example is not work , how can I fix it?

CaffeMrDe, 2021-04-10:
ROS_MASTER_URI=http://localhost:11311

process[virtual_joint_broadcaster_0-1]: started with pid [69302]
process[online_planning_example_node-2]: started with pid [69303]
process[online_planning_example_rviz-3]: started with pid [69309]
process[joint_state_publisher_gui-4]: started with pid [69310]
You can set logging level with TRAJOPT_LOG_THRESH. Valid values: FATAL ERROR WARN INFO DEBUG TRACE. Defaulting to ERROR
Warning: SRDF Parser: No version number was provided so latest parser (version: 1.0.0) will be used.
         at line 89 in /home/fshs/tesseract_ros_ws/src/tesseract/tesseract_scene_graph/src/srdf_model.cpp
[ERROR] [1618032740.518226571]: Press enter to setup the problem
[ WARN] [1618032746.440352934]: Link gantry_base has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1618032746.440552464]: Link gantry_axis_1 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1618032746.440626698]: Link gantry_axis_2 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1618032746.503287438]: No kinematics plugins defined. Fill and load kinematics.yaml!

Target Joint Position: 5.5   3   0   0   0   0   0   0
Target TF:
-2.65359e-06            1  0.000796327        5.703
 2.11802e-09 -0.000796327            1      4.85984
           1  2.65359e-06 -4.89664e-12        0.545
           0            0            0            1

************************************************************
    IFOPT - Interface to Nonlinear Optimizers (v2.0)
                © Alexander W. Winkler
           https://github.com/ethz-adrl/ifopt
************************************************************
Legend:
c - number of variables, constraints or cost terms
i - indices of this set in overall problem
v - number of [violated variable- or constraint-bounds] or [cost term value]

                                   c             i             v 
variable-sets:
   Joint_Position_0                 8        0......7           0
   Joint_Position_1                 8        8.....15           0
   Joint_Position_2                 8       16.....23           0
   Joint_Position_3                 8       24.....31           0
   Joint_Position_4                 8       32.....39           0
   Joint_Position_5                 8       40.....47           0
   Joint_Position_6                 8       48.....55           0
   Joint_Position_7                 8       56.....63           0
   Joint_Position_8                 8       64.....71           0
   Joint_Position_9                 8       72.....79           0
   Joint_Position_10                8       80.....87           0
   Joint_Position_11                8       88.....95           0

constraint-sets:
   Home_Position                    8        0......7           0
   CartPos                          6        8.....13           0

cost-terms:
   JointVelocity_squared_cost       1        0......0        3.57
   Collision                        1        0......0           0
   Collision                        1        1......1           0
   Collision                        1        2......2           0
   Collision                        1        3......3           0
   Collision                        1        4......4           1
   Collision                        1        5......5           1
   Collision                        1        6......6           1
   Collision                        1        7......7           1
   Collision                        1        8......8           1
   Collision                        1        9......9           1
   Collision                        1       10.....10           1

[ERROR] [1618032753.381541670]: Press enter to run live
[ERROR] [1618032753.381644633]: Then use the joint state publisher gui to move the human_x/y joints or the target_x/y/z joints

Solver Frequency (Hz): 33.206043
Cost: 4.143267
[ERROR] [1618032775.365156715]: No environment
Solver Frequency (Hz): 31.990787
Cost: 4.141348
[ERROR] [1618032775.393988895]: No environment
Solver Frequency (Hz): 32.454888
Cost: 4.139500
[ERROR] [1618032775.437327347]: No environment
Solver Frequency (Hz): 33.909800
Cost: 4.137598
[ERROR] [1618032775.480172552]: No environment
Solver Frequency (Hz): 32.583904
Cost: 4.135574
Solver Frequency (Hz): 33.044743
Cost: 4.133767
[ERROR] [1618032775.530371386]: No environment
[ERROR] [1618032775.530755206]: No environment
Solver Frequency (Hz): 32.822398
Cost: 4.132078
[ERROR] [1618032775.568681271]: No environment
Solver Frequency (Hz): 31.992834
Cost: 4.130309
[ERROR] [1618032775.588419562]: No environment
Solver Frequency (Hz): 31.723875
Cost: 4.128469
[ERROR] [1618032775.631714503]: No environment
Solver Frequency (Hz): 32.672265
Cost: 4.126717
[ERROR] [1618032775.675683719]: No environment
Solver Frequency (Hz): 34.286498
Cost: 4.125072
Solver Frequency (Hz): 33.307797
Cost: 4.123360
[ERROR] [1618032775.720766659]: No environment
[ERROR] [1618032775.720946434]: No environment
Solver Frequency (Hz): 31.970332
Cost: 4.121590
[ERROR] [1618032775.763881260]: No environment

CaffeMrDe, 2021-04-10:
[image]

CaffeMrDe, 2021-04-10:
the scense is not exist. But it look seems normal.

Levi-Armstrong, 2021-04-12:
In the case of the online example you need to switch the display mode to URDF. This is the only example which does not use a environment monitor. I have someone who should be submitting a PR soon which updated this example to use a different Rviz config.

CaffeMrDe, 2021-04-12:
Ok， I see.Many thanks for your kind and warm reply. I work it . And When I  run it on Virtual machine without GPU ,Solver Frequency (Hz) is 33.909800, which it looks slow .  (I have alreay adjust the box size , But it can't speed solvering up). Can  I speed up the solver?

Levi-Armstrong, 2021-04-12:
It may be built in debug, the following command will build everything in release `catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release`.

CaffeMrDe, 2021-04-14:
Ok, I try. It work. It get 100hz . You are excellent.

closed 2021-04-14 by CaffeMrDe

Levi-Armstrong, 2021-04-14:
That is great to hear! So just for reference point I am running a HP Z4 Workstation with Intel Core i9 processors and 30GB of RAM and can get around 600hz.