tesseract-robotics/tesseract_ros #112: Build Error: stdlib.h: no such file or directory

ben-greenberg, 2021-05-25:
I ran into this strange build issue where the packages are unable to find `stdlib.h`. I found some discussion on another ROS issue that suggested omitting the `SYSTEM` from the `target_include_directories` calls. This seems to fix the issue for me. Not sure if its a bad idea to merge it into master. 

https://github.com/ros-industrial-consortium/tesseract_ros/blob/4c5cbdb2938dafd96c050c74b5501a3e7c52fd13/tesseract_plugins/CMakeLists.txt#L38

https://github.com/bulwahn/image_common/commit/31baf37689faa1d8b5d1f73d9847b757bd7085b2

```
Errors     << tesseract_monitoring:make /home/ros-industrial/workspaces/proactive_ws/logs/tesseract_monitoring/build.make.000.log                                                                         
In file included from /usr/include/c++/9/ext/string_conversions.h:41,
                 from /usr/include/c++/9/bits/basic_string.h:6493,
                 from /usr/include/c++/9/string:55,
                 from /usr/include/c++/9/bits/locale_classes.h:40,
                 from /usr/include/c++/9/bits/ios_base.h:41,
                 from /usr/include/c++/9/ios:42,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /opt/ros/noetic/include/ros/console.h:38,
                 from /home/ros-industrial/workspaces/proactive_ws/src/tesseract_ros/tesseract_monitoring/src/environment_monitor_interface.cpp:28:
/usr/include/c++/9/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
   75 | #include_next <stdlib.h>
      |               ^~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/tesseract_monitoring_environment_interface.dir/build.make:63: CMakeFiles/tesseract_monitoring_environment_interface.dir/src/environment_monitor_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/tesseract_monitoring_environment_interface.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/include/c++/9/ext/string_conversions.h:41,
                 from /usr/include/c++/9/bits/basic_string.h:6493,
                 from /usr/include/c++/9/string:55,
                 from /usr/include/c++/9/bits/locale_classes.h:40,
                 from /usr/include/c++/9/bits/ios_base.h:41,
                 from /usr/include/c++/9/ios:42,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /opt/ros/noetic/include/ros/console.h:38,
                 from /home/ros-industrial/workspaces/proactive_ws/src/tesseract_ros/tesseract_monitoring/src/environment_monitor.cpp:39:
/usr/include/c++/9/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
   75 | #include_next <stdlib.h>
      |               ^~~~~~~~~~
compilation terminated.
In file included from /usr/include/c++/9/ext/string_conversions.h:41,
                 from /usr/include/c++/9/bits/basic_string.h:6493,
                 from /usr/include/c++/9/string:55,
                 from /usr/include/c++/9/bits/locale_classes.h:40,
                 from /usr/include/c++/9/bits/ios_base.h:41,
                 from /usr/include/c++/9/ios:42,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /opt/ros/noetic/include/ros/console.h:38,
                 from /home/ros-industrial/workspaces/proactive_ws/src/tesseract_ros/tesseract_monitoring/src/current_state_monitor.cpp:39:
/usr/include/c++/9/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
   75 | #include_next <stdlib.h>
      |               ^~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/tesseract_monitoring_environment.dir/build.make:76: CMakeFiles/tesseract_monitoring_environment.dir/src/current_state_monitor.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/tesseract_monitoring_environment.dir/build.make:63: CMakeFiles/tesseract_monitoring_environment.dir/src/environment_monitor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:184: CMakeFiles/tesseract_monitoring_environment.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

Levi-Armstrong, 2022-04-02:
@ben-greenberg Is this still an issue?

Levi-Armstrong, 2023-02-28:
Reopen if still an issue

closed 2023-02-28 by Levi-Armstrong

DavidMerzJr, 2024-03-28:
@marip8 @marrts @Levi-Armstrong

@collinthornton is experiencing this issue as well.  We have a relatively recent Tesseract build working on Ubuntu 18.04. When trying to build that same workspace in Noetic, tesseract_rosutils and tesseract_monitoring are both failing. Ben's action fixes this in our case as well. Was the underlying issue here ever solved?

Levi-Armstrong, 2024-03-30:
For those having this issue, if you search your system for this file where is it located?

collinthornton, 2024-04-01:
The source file is located in `~/projects/wkspc/src/tesseract_deps/tesseract_ros/tesseract_rosutils/CMakeLists.tx`

Levi-Armstrong, 2024-04-01:
I should have been more specific. Interested in the location of this file stdlib.h.

collinthornton, 2024-04-01:
[image]

Levi-Armstrong, 2024-04-02:
I searched the internet a little and it suggest that this is an environment issues since the file is provided by the standard and should always be found.

Levi-Armstrong, 2024-04-03:
https://lists.debian.org/debian-glibc/2020/04/msg00021.html