tesseract-robotics/tesseract_ros #121: Run tesseract_ros tutorial

gautz, 2021-07-06:
I am trying to run the tuto for Tesseract under ROS: https://tesseract-docs.readthedocs.io/en/latest/_source/ros/overview/tesseract_roscpp_interface_doc.html

I setup my workspace following:
https://tesseract-docs.readthedocs.io/en/latest/_source/intro/getting_started_doc.html
Here are some issues, did I do something wrong? :
- I had to update the dependencies to match https://github.com/ros-industrial-consortium/tesseract_ros/blob/master/dependencies.rosinstall
- tesseract_rviz build is failing
```
Errors     << tesseract_rviz:make /home/gah/tesseract_ws/logs/tesseract_rviz/build.make.004.log                   
/home/gah/tesseract_ws/src/tesseract_ros/tesseract_rviz/src/render_tools/environment_widget.cpp: In member function ‘void tesseract_rviz::EnvironmentWidget::onUpdate()’:
/home/gah/tesseract_ws/src/tesseract_ros/tesseract_rviz/src/render_tools/environment_widget.cpp:183:55: error: ‘using element_type = class tesseract_environment::Environment’ {aka ‘class tesseract_environment::Environment’} has no member named ‘getCurrentStateTimestamp’
  183 |     if (update_required_ || state_timestamp_ != env_->getCurrentStateTimestamp())
      |                                                       ^~~~~~~~~~~~~~~~~~~~~~~~
/home/gah/tesseract_ws/src/tesseract_ros/tesseract_rviz/src/render_tools/environment_widget.cpp:187:32: error: ‘using element_type = class tesseract_environment::Environment’ {aka ‘class tesseract_environment::Environment’} has no member named ‘getCurrentStateTimestamp’
  187 |       state_timestamp_ = env_->getCurrentStateTimestamp();
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/tesseract_rviz_render_tools.dir/build.make:189: CMakeFiles/tesseract_rviz_render_tools.dir/src/render_tools/environment_widget.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:368: CMakeFiles/tesseract_rviz_render_tools.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/gah/tesseract_ws/build/tesseract_rviz; catkin build --get-env tesseract_rviz | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
```

gautz, 2021-07-07:
when building without tesseract_ext (https://github.com/ros-industrial-consortium/tesseract_ros/blob/master/dependencies.rosinstall) I get
```
Errors     << trajopt_ifopt:make /home/gah/tesseract_ros_workcell/logs/trajopt_ifopt/build.make.000.log           
/home/gah/tesseract_ros_workcell/src/trajopt/trajopt_ifopt/src/squared_cost.cpp: In member function ‘virtual void trajopt::SquaredCost::FillJacobianBlock(std::string, ifopt::Component::Jacobian&) const’:
/home/gah/tesseract_ros_workcell/src/trajopt/trajopt_ifopt/src/squared_cost.cpp:76:56: error: ‘virtual void ifopt::ConstraintSet::FillJacobianBlock(std::string, ifopt::Component::Jacobian&) const’ is private within this context
   76 |   constraint_->FillJacobianBlock(var_set, cnt_jac_block);
      |                                                        ^
In file included from /opt/ros/noetic/include/ifopt/cost_term.h:33,
                 from /home/gah/tesseract_ros_workcell/src/trajopt/trajopt_ifopt/include/trajopt_ifopt/costs/squared_cost.h:31,
                 from /home/gah/tesseract_ros_workcell/src/trajopt/trajopt_ifopt/src/squared_cost.cpp:26:
/opt/ros/noetic/include/ifopt/constraint_set.h:109:16: note: declared private here
  109 |   virtual void FillJacobianBlock(std::string var_set, Jacobian& jac_block) const = 0;
      |                ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/trajopt_ifopt.dir/build.make:141: CMakeFiles/trajopt_ifopt.dir/src/squared_cost.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/trajopt_ifopt.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
cd /home/gah/tesseract_ros_workcell/build/trajopt_ifopt; catkin build --get-env trajopt_ifopt | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
```

mpowelson, 2021-07-07:
We made some changes to IFOPT to support these functions that I don't think are in the released version. Did wstool pull the correct version (27451ce3212d3c0ccd3a0db9a1ee9a9963466674)? That's your second issue.

For your first issue, it looks like it might be a version issue as well. I'm not sure if those functions have changed recently, but it seems likely given your second issue.

@steviedale might be interested in this.

gautz, 2021-07-07:
- Am I supposed to use https://github.com/ros-industrial-consortium/tesseract_ros/blob/master/dependencies_with_ext.rosinstall or /dependencies.rosinstall ? Or stay on the master branch for all packages that are not on a specific tag?
- You mean I should checkout 27451ce for ifopt? Tried it and it did not work

I guess the easiest way would be to add a dependencies.rosinstall in tesseract_ros_workcell as suggested here:
https://github.com/Levi-Armstrong/tesseract_ros_workcell/issues/4

gautz, 2021-09-03:
somehow I tried again and it worked with dependencies.rosinstall

closed 2021-09-03 by gautz