tesseract-robotics/tesseract_ros #153: Enabling Trajopt debug messages when using launch files

cpetersmeier, 2022-05-13:
Hello everyone,

is there a way to get trajopt debug output when using the launch file, rather than running the binary?
I get verbose output when running:

    workspace/devel/bin$ ./tesseract_examples_basic_cartesian_example

However, I do not get any output when using the launch file:

    roslaunch tesseract_ros_examples basic_cartesian_example.launch 

I made sure I set the following in my bashrc:

    export TRAJOPT_LOG_THRESH=DEBUG

Thanks in advance!

Levi-Armstrong, 2022-05-13:
I believe this controlled by setting the console bridge log level to debug. Not sure if there is a way to set this outside of c++. @mpowelson do you know if it is possible?

``` c++
if (debug_)
    console_bridge::setLogLevel(console_bridge::LogLevel::CONSOLE_BRIDGE_LOG_DEBUG);
```

rhaschke, 2022-05-17:
You need to use the [roslaunch node attribute `output="screen"`](http://wiki.ros.org/roslaunch/XML/node) to see stdout/DEBUG messages on the screen. Otherwise they just go into a log _file_.

closed 2022-05-17 by cpetersmeier