tesseract-robotics/tesseract_ros #161: is the demo about": https://rosindustrial.org/news/2022/2/21/using-tesseract-and-trajopt-for-a-real-time-application" has been open-sourced

dbdxnuliba, 2022-07-28:
[image]
is the demo about": https://rosindustrial.org/news/2022/2/21/using-tesseract-and-trajopt-for-a-real-time-application" has been open-sourced,and  where we can find it and have a  try ,thanks.

Levi-Armstrong, 2022-07-28:
@ben-greenberg Is this publicly available? @dbdxnuliba This was build on the basic implementation of the online_planning_example found in both tesseract_examples and tesseract_ros_examples.

dbdxnuliba, 2022-07-29:
thanks,I see, online planning demo is only one 
[image]
 I hope there will be much demo opensourced in the future,to make use of online planning provided by tesseract

ben-greenberg, 2022-07-30:
Unfortunately, the application referenced is not publicly available at this point.

I will echo your suggestion to use the online_planning_example as a starting point for new application development.

You might also check out [this](https://github.com/ros-industrial-consortium/scan_n_plan_workshop)  project for another example of a fully developed application that leverages Tesseract.

dbdxnuliba, 2022-08-01:
I see ,thanks a lot