tesseract-robotics/tesseract_ros #162: I use franka  urdf and srdf to test tesseract

dbdxnuliba, 2022-08-01:
roslaunch tesseract_ros_examples basic_cartesian_example_franka.launch
[image]

and I USE  urdf from  pkg:https://github.com/frankaemika/franka_ros/blob/melodic-devel/franka_description/robots/panda_arm.urdf.xacro 
and srdf from https://github.com/ros-planning/panda_moveit_config/blob/melodic-devel/config/panda_arm.srdf.xacro
like that:

**basic_cartesian_example_franka.launch file content:**
`<?xml version="1.0"?>
<launch>
  
  
<arg name="load_gripper" default="false" />

  <arg name="robot_description" default="robot_description"/>
  <arg name="trajopt_description" default="trajopt_description"/>
  <arg name="plotting" default="true"/>
  <arg name="rviz" default="true"/>
  <arg name="ifopt" default="false"/>
  <arg name="debug" default="false"/>
  <arg name="testing" default="false"/>

  
  <param name="$(arg robot_description)" command="$(find xacro)/xacro --inorder '$(find franka_description)/robots/panda_arm.urdf.xacro'" /> 
  

  

  <param name="$(arg robot_description)_semantic" command="$(find xacro)/xacro '$(find panda_moveit_config)/config/panda_arm_tesseract.srdf'" />

  <group unless="$(arg testing)">
    
    <node pkg="tf" type="static_transform_publisher" name="virtual_joint_broadcaster_0" args="0 0 0 0 0 0 world panda_link0 100" />

    <node pkg="tesseract_ros_examples" type="tesseract_ros_examples_basic_cartesian_example_franka_node" name="tesseract_ros_examples_basic_cartesian_example_franka_node" output="screen">
      <param name="plotting" type="bool" value="$(arg plotting)"/>
      <param name="rviz" type="bool" value="$(arg rviz)"/>
      <param name="ifopt" type="bool" value="$(arg ifopt)"/>
      <param name="debug" type="bool" value="$(arg debug)"/>
    </node>

    
    <node if="$(arg rviz)" pkg="rviz" type="rviz" name="tesseract_ros_examples_basic_cartesian_example_rviz"
         args="-d $(find tesseract_ros_examples)/config/examples_franka_produce.rviz" />
  </group>
 </launch>`

**and  basic_cartesian_example_franka.cpp like that:**
`/**
 * @file basic_cartesian_plan.cpp
 * @brief Basic cartesian example implementation
 *
 * @author Levi Armstrong
 * @date July 22, 2019
 * @version TODO
 * @bug No known bugs
 *
 * @copyright Copyright (c) 2017, Southwest Research Institute
 *
 * @par License
 * Software License Agreement (Apache License)
 * @par
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 * http://www.apache.org/licenses/LICENSE-2.0
 * @par
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
[... 221 lines cut ...]
  tesseract_common::Timer stopwatch;
  stopwatch.start();
  ProcessPlanningFuture response = planning_server.run(request);
  planning_server.waitForAll();
  stopwatch.stop();
  CONSOLE_BRIDGE_logInform("Planning took %f seconds.", stopwatch.elapsedSeconds());

  // Plot Process Trajectory
  if (plotter_ != nullptr && plotter_->isConnected())
  {
    plotter_->waitForInput();
    const auto& ci = response.problem->results->as<CompositeInstruction>();
    tesseract_common::Toolpath toolpath = toToolpath(ci, *env_);
    tesseract_common::JointTrajectory trajectory = toJointTrajectory(ci);
    auto state_solver = env_->getStateSolver();
    plotter_->plotMarker(ToolpathMarker(toolpath));
    plotter_->plotTrajectory(trajectory, *state_solver);
  }

  CONSOLE_BRIDGE_logInform("Final trajectory is collision free");
  return response.interface->isSuccessful();
}

}  // namespace tesseract_examples
`

dbdxnuliba, 2022-08-01:
<?xml version="1.0" ?>
<robot name="panda">
    
    
    
    
    
    <group name="panda_arm">
      <chain base_link="panda_link0" tip_link="panda_link8" />
    </group>

    <group name="panda_arm_joint_group">
        <joint name="panda_joint1"/>
        <joint name="panda_joint2"/>
        <joint name="panda_joint3"/>
        <joint name="panda_joint4"/>
        <joint name="panda_joint5"/>
        <joint name="panda_joint6"/>
        <joint name="panda_joint7"/>
    </group>

    <group_state name="all-zeros" group="panda_arm">
        <joint name="panda_joint1" value="0" />
        <joint name="panda_joint2" value="0" />
        <joint name="panda_joint3" value="0" />
        <joint name="panda_joint4" value="0" />
        <joint name="panda_joint5" value="0" />
        <joint name="panda_joint6" value="0" />
        <joint name="panda_joint7" value="0" />
    </group_state>

    <group_tcps group="panda_arm">
      <tcp name="laser" xyz="1 .1 1" rpy="0 1.57 0" />
      <tcp name="welder" xyz=".1 1 .2" wxyz="1 0 0 0" />
    </group_tcps>

    <group_tcps group="panda_arm_joint_group">
      <tcp name="laser" xyz="1 .1 1" rpy="0 1.57 0" />
      <tcp name="welder" xyz=".1 1 .2" wxyz="1 0 0 0" />
    </group_tcps>

        
    <kinematics_plugin_config filename="package://tesseract_support/urdf/franka_plugins.yaml"/>

    
    <contact_managers_plugin_config filename="package://tesseract_support/urdf/contact_manager_plugins.yaml"/>

    
    <group_state name="ready" group="panda_arm">
      <joint name="panda_joint1" value="0" />
      <joint name="panda_joint2" value="-0.785" />
      <joint name="panda_joint3" value="0" />
      <joint name="panda_joint4" value="-2.356" />
      <joint name="panda_joint5" value="0" />
      <joint name="panda_joint6" value="1.571" />
      <joint name="panda_joint7" value="0.785" />
    </group_state>
    <group_state name="extended" group="panda_arm">
      <joint name="panda_joint1" value="0" />
      <joint name="panda_joint2" value="0" />
      <joint name="panda_joint3" value="0" />
      <joint name="panda_joint4" value="0" />
      <joint name="panda_joint5" value="0" />
      <joint name="panda_joint6" value="0" />
      <joint name="panda_joint7" value="0.785" />
    </group_state>
    <group_state name="transport" group="panda_arm">
      <joint name="panda_joint1" value="0" />
      <joint name="panda_joint2" value="-0.5599" />
      <joint name="panda_joint3" value="0" />
      <joint name="panda_joint4" value="-2.97" />
      <joint name="panda_joint5" value="0" />
      <joint name="panda_joint6" value="0" />
      <joint name="panda_joint7" value="0.785" />
    </group_state>
    
    
    
    
    <disable_collisions link1="panda_link0" link2="panda_link1" reason="Adjacent" />
    <disable_collisions link1="panda_link0" link2="panda_link2" reason="Never" />
    <disable_collisions link1="panda_link0" link2="panda_link3" reason="Never" />
    <disable_collisions link1="panda_link0" link2="panda_link4" reason="Never" />
    <disable_collisions link1="panda_link1" link2="panda_link2" reason="Adjacent" />
    <disable_collisions link1="panda_link1" link2="panda_link3" reason="Default" />
    <disable_collisions link1="panda_link1" link2="panda_link4" reason="Never" />
    <disable_collisions link1="panda_link2" link2="panda_link3" reason="Adjacent" />
    <disable_collisions link1="panda_link2" link2="panda_link4" reason="Never" />
    <disable_collisions link1="panda_link3" link2="panda_link4" reason="Adjacent" />
    <disable_collisions link1="panda_link3" link2="panda_link6" reason="Never" />
    <disable_collisions link1="panda_link4" link2="panda_link5" reason="Adjacent" />
    <disable_collisions link1="panda_link4" link2="panda_link6" reason="Never" />
    <disable_collisions link1="panda_link4" link2="panda_link7" reason="Never" />
    <disable_collisions link1="panda_link4" link2="panda_link8" reason="Never" />
    <disable_collisions link1="panda_link5" link2="panda_link6" reason="Adjacent" />
    <disable_collisions link1="panda_link6" link2="panda_link7" reason="Adjacent" />
    <disable_collisions link1="panda_link6" link2="panda_link8" reason="Default" />
    <disable_collisions link1="panda_link7" link2="panda_link8" reason="Adjacent" />
</robot>

dbdxnuliba, 2022-08-01:
**and panda_arm.xacro like that:**

<?xml version="1.0" ?>
<robot name="panda">
    
    
    
    
    
    <group name="panda_arm">
      <chain base_link="panda_link0" tip_link="panda_link8" />
    </group>

    <group name="panda_arm_joint_group">
        <joint name="panda_joint1"/>
        <joint name="panda_joint2"/>
        <joint name="panda_joint3"/>
        <joint name="panda_joint4"/>
        <joint name="panda_joint5"/>
        <joint name="panda_joint6"/>
        <joint name="panda_joint7"/>
    </group>

    <group_state name="all-zeros" group="panda_arm">
        <joint name="panda_joint1" value="0" />
        <joint name="panda_joint2" value="0" />
        <joint name="panda_joint3" value="0" />
        <joint name="panda_joint4" value="0" />
        <joint name="panda_joint5" value="0" />
        <joint name="panda_joint6" value="0" />
        <joint name="panda_joint7" value="0" />
    </group_state>

    <group_tcps group="panda_arm">
      <tcp name="laser" xyz="1 .1 1" rpy="0 1.57 0" />
      <tcp name="welder" xyz=".1 1 .2" wxyz="1 0 0 0" />
    </group_tcps>

    <group_tcps group="panda_arm_joint_group">
      <tcp name="laser" xyz="1 .1 1" rpy="0 1.57 0" />
      <tcp name="welder" xyz=".1 1 .2" wxyz="1 0 0 0" />
    </group_tcps>

        
    <kinematics_plugin_config filename="package://tesseract_support/urdf/franka_plugins.yaml"/>

    
    <contact_managers_plugin_config filename="package://tesseract_support/urdf/contact_manager_plugins.yaml"/>

    
    <group_state name="ready" group="panda_arm">
      <joint name="panda_joint1" value="0" />
      <joint name="panda_joint2" value="-0.785" />
      <joint name="panda_joint3" value="0" />
      <joint name="panda_joint4" value="-2.356" />
      <joint name="panda_joint5" value="0" />
      <joint name="panda_joint6" value="1.571" />
      <joint name="panda_joint7" value="0.785" />
    </group_state>
    <group_state name="extended" group="panda_arm">
      <joint name="panda_joint1" value="0" />
      <joint name="panda_joint2" value="0" />
      <joint name="panda_joint3" value="0" />
      <joint name="panda_joint4" value="0" />
      <joint name="panda_joint5" value="0" />
      <joint name="panda_joint6" value="0" />
      <joint name="panda_joint7" value="0.785" />
    </group_state>
    <group_state name="transport" group="panda_arm">
      <joint name="panda_joint1" value="0" />
      <joint name="panda_joint2" value="-0.5599" />
      <joint name="panda_joint3" value="0" />
      <joint name="panda_joint4" value="-2.97" />
      <joint name="panda_joint5" value="0" />
      <joint name="panda_joint6" value="0" />
      <joint name="panda_joint7" value="0.785" />
    </group_state>
    
    
    
    
    <disable_collisions link1="panda_link0" link2="panda_link1" reason="Adjacent" />
    <disable_collisions link1="panda_link0" link2="panda_link2" reason="Never" />
    <disable_collisions link1="panda_link0" link2="panda_link3" reason="Never" />
    <disable_collisions link1="panda_link0" link2="panda_link4" reason="Never" />
    <disable_collisions link1="panda_link1" link2="panda_link2" reason="Adjacent" />
    <disable_collisions link1="panda_link1" link2="panda_link3" reason="Default" />
    <disable_collisions link1="panda_link1" link2="panda_link4" reason="Never" />
    <disable_collisions link1="panda_link2" link2="panda_link3" reason="Adjacent" />
    <disable_collisions link1="panda_link2" link2="panda_link4" reason="Never" />
    <disable_collisions link1="panda_link3" link2="panda_link4" reason="Adjacent" />
    <disable_collisions link1="panda_link3" link2="panda_link6" reason="Never" />
    <disable_collisions link1="panda_link4" link2="panda_link5" reason="Adjacent" />
    <disable_collisions link1="panda_link4" link2="panda_link6" reason="Never" />
    <disable_collisions link1="panda_link4" link2="panda_link7" reason="Never" />
    <disable_collisions link1="panda_link4" link2="panda_link8" reason="Never" />
    <disable_collisions link1="panda_link5" link2="panda_link6" reason="Adjacent" />
    <disable_collisions link1="panda_link6" link2="panda_link7" reason="Adjacent" />
    <disable_collisions link1="panda_link6" link2="panda_link8" reason="Default" />
    <disable_collisions link1="panda_link7" link2="panda_link8" reason="Adjacent" />
</robot>

dbdxnuliba, 2022-08-01:
can you help me to point out the reason why tesseract_rviz/TesseractWorkbench crash  when I use panda urdf and srdf

Levi-Armstrong, 2022-08-01:
@dbdxnuliba I believe PR #163 should solve the issue. Let me know if it resolves the issue.

dbdxnuliba, 2022-08-02:
thanks , your update version solve the issue,it works well now.
[image]

closed 2022-08-02 by Levi-Armstrong