tesseract-robotics/tesseract_ros #164: could you provide demo about how to run  on a real robot in the near future

dbdxnuliba, 2022-08-02:
could you provide demo about how to run  on a real robot in the near future

Levi-Armstrong, 2022-08-03:
Depending on your robot, but I would say that the common robots used for research have a ros driver which has an FollowJointTrajectoryAction which you can fill out with the trajectory from the motion planning for executions.

dbdxnuliba, 2022-08-04:
I found the plan result of tesseract from basic_cartesian_example isthe topic  /tesseract/display_tesseract_trajectory, but not FollowJointTrajectoryAction , is that right?
if I want to use FollowJointTrajectoryAction ,where can I find the interface about FollowJointTrajectoryAction   in tesseract project

Levi-Armstrong, 2022-08-04:
No, what robot are you wanting to execute this on?

dbdxnuliba, 2022-08-05:
franka robot