tesseract-robotics/tesseract_ros #170: how  can we import 3d pointcloud topic to  plansence in tesseract_ros，

dbdxnuliba, 2022-08-11:
how  can we import 3d pointcloud topic  to plansence。

We know  we can import the point cloud into  moveit convinently by 
config  the file
sensor_manager.launch.xml to produce octomap itself in moveit,
how about tesseract_ros to deal with it.

Levi-Armstrong, 2022-08-11:
Right now I currently do this by using the octomap and creating an add link command  which gets provided as an additional command to the planning request. The long term plan is to update the planning server to support providing a topic to monitor  and maintain an internal octomap. Where if this exist it would automatically get added to incoming planning request. This would not take much to implement, but have not gotten around to doing it. If interested in tackling this I would happily review the PR.

dbdxnuliba, 2022-08-12:
yes ,I'm interested in the function