tesseract-robotics/tesseract_ros #172: how can we combine tesseract  with moveit  in usage

dbdxnuliba, 2022-09-02:
how can we combine tesseract  with moveit  in usage

dbdxnuliba, 2022-09-06:
such as action_client and octomap  planning scene  proviede by  moveit, could we provide interface about them

Levi-Armstrong, 2022-09-06:
Your just wanting an integrated octomap in the planning server where the octomap is added during motion planning request?

dbdxnuliba, 2022-09-09:
yes it is.

Levi-Armstrong, 2022-09-09:
Okay, this should be easy to implement. I will put something together.

dbdxnuliba, 2022-09-13:
thanks a lot for your hard work

dbdxnuliba, 2022-09-15:
> Okay, this should be easy to implement. I will put something together.

please tell me ,when you add the function ,thanks a lot

dbdxnuliba, 2022-09-15:
and my D435i rgbd camera mounted on arm_end_effector, has been calibrated ,so I can publish the  pointcloud topic,so I want an integrated octomap based on pointcloud topic  in the planning server where the octomap is added during motion planning request?