tesseract-robotics/tesseract_ros #176: INFO] optimization couldn't satisfy all constraints when roslaunch tesseract_ros_examples puzzle_piece_auxillary_axes_example.launch

dbdxnuliba, 2022-09-19:
roslaunch tesseract_ros_examples puzzle_piece_auxillary_axes_example.launch 

INFO] optimization couldn't satisfy all constraints
Info:    TRAJOPT motion planning failed (Failed to find valid solution) for process input: Tesseract Composite Instruction
Debug:   worker 4 finished running TRAJOPT
Debug:   worker 0 ready to run Error Callback
[image]

Levi-Armstrong, 2022-09-19:
Are you running 0.13.X version?

dbdxnuliba, 2022-09-20:
it's 0.12.0

Levi-Armstrong, 2022-09-20:
Interesting. I will see if I can track down the release where it broke.

Levi-Armstrong, 2022-09-21:
This was broken when we switched OSQP to enable adaptive_rho by default which in most cases speeds up convergences, but in this case it prevents it from converging. I am exposing the convex solver config in the trajopt solver profile so this can be disabled.

Levi-Armstrong, 2022-09-26:
Fixed by recent PR.

closed 2022-09-26 by Levi-Armstrong

Levi-Armstrong, 2022-10-20:
> where can I modify the config param
> 
> > This was broken when we switched OSQP to enable adaptive_rho by default which in most cases speeds up convergences, but in this case it prevents it from converging. I am exposing the convex solver config in the trajopt solver profile so this can be disabled.

Here is what I had to modify

https://github.com/tesseract-robotics/tesseract_planning/blob/00726c502b3a71c2df02576e0012037884e4ef6a/tesseract_examples/src/puzzle_piece_auxillary_axes_example.cpp#L211-L218