tesseract-robotics/tesseract_ros #181: catkin build err,tesseract_planning-0.13.0,tesseract_planning-0.13.0/tesseract_motion_planners/descartes/src/profile/descartes_default_plan_profile.cpp:31:16:   required from here /usr/include/c++/9/ext/new_allocator.h:146:4: error: invalid new-expression of abstract class type 'descartes_light::StateEvaluator<float>'

dbdxnuliba, 2022-09-23:
my env:
ubuntu20.04 
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1) 
catkin build -j3
pkg :tesseract_planning-0.13.0

/home/taimi/ws_tesseract_060/src/tesseract_planning-0.13.0/tesseract_motion_planners/descartes/src/profile/descartes_default_plan_profile.cpp:31:16:   required from here
/usr/include/c++/9/ext/new_allocator.h:146:4: error: invalid new-expression of abstract class type 'descartes_light::StateEvaluator<float>'
  146 |  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
[image]

Levi-Armstrong, 2022-09-23:
Also if you are building tesseract from source there is a know issue with having it install as binary. So I would remove ros-noetic-tesseract*

vegfood, 2022-10-13:
> my env: ubuntu20.04 gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1) catkin build -j3 pkg :tesseract_planning-0.13.0
> 
> /home/taimi/ws_tesseract_060/src/tesseract_planning-0.13.0/tesseract_motion_planners/descartes/src/profile/descartes_default_plan_profile.cpp:31:16: required from here /usr/include/c++/9/ext/new_allocator.h:146:4: error: invalid new-expression of abstract class type 'descartes_light::StateEvaluator' 146 | { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); } [image]

I have encountered the same problem as mentioned above with descartes_light version 0.3.0. After I checkout current master HEAD of descates_light (rev. fbb74ebe679ba21cf1a49d4f7b5e353da75f0549), the error disappears.

dbdxnuliba, 2022-10-19:
> Can you provide the versions of descates, tesseract, tesseract_planning, tesseract_ros, tesseract_qt that you are using?

descartes_light version 0.3.0,tesseract-0.13.1,tesseract_planning-0.13.0,tesseract_ros0.6.0,tesseract_qt-0.13.0

dbdxnuliba, 2022-10-19:
@vegfood ,yes it is, we fix the issue by using descartes_light current master HEAD of descates_light (rev. fbb74ebe679ba21cf1a49d4f7b5e353da75f0549),

> I have encountered the same problem as mentioned above with descartes_light version 0.3.0. After I checkout current master HEAD of descates_light (rev. fbb74ebe679ba21cf1a49d4f7b5e353da75f0549), the [...]

dbdxnuliba, 2022-10-24:
you can close the issue now thanks

closed 2022-10-24 by Levi-Armstrong