tesseract-robotics/tesseract_ros #195: Unexpected crash when running the demo

sleichu2, 2023-02-07:
Error running demo
tesseract_ros version:master

<pre>roslaunch tesseract_ros_examples basic_cartesian_example.launch 
... logging to /home/rikka/.ros/log/0b94801a-a6c1-11ed-baa5-0dd8e31ec9e6/roslaunch-rnc-1559010.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is &lt;1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
<b>started roslaunch server http://rnc:43355/</b>

SUMMARY
========

PARAMETERS
 * /robot_description: &lt;?xml version=&quot;1....
 * /robot_description_semantic: &lt;?xml version=&quot;1....
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /tesseract_ros_examples_basic_cartesian_example_node/debug: False
 * /tesseract_ros_examples_basic_cartesian_example_node/ifopt: False
 * /tesseract_ros_examples_basic_cartesian_example_node/plotting: True
 * /tesseract_ros_examples_basic_cartesian_example_node/rviz: True

NODES
  /
    tesseract_ros_examples_basic_cartesian_example_node (tesseract_ros_examples/tesseract_ros_examples_basic_cartesian_example_node)
    tesseract_ros_examples_basic_cartesian_example_rviz (rviz/rviz)
    virtual_joint_broadcaster_0 (tf/static_transform_publisher)

auto-starting new master
<b>process[master]: started with pid [1559024]</b>
<b>ROS_MASTER_URI=http://localhost:11311</b>

<b>setting /run_id to 0b94801a-a6c1-11ed-baa5-0dd8e31ec9e6</b>
<b>process[rosout-1]: started with pid [1559035]</b>
started core service [/rosout]
<b>process[virtual_joint_broadcaster_0-2]: started with pid [1559042]</b>
<b>process[tesseract_ros_examples_basic_cartesian_example_node-3]: started with pid [1559043]</b>
<b>process[tesseract_ros_examples_basic_cartesian_example_rviz-4]: started with pid [1559044]</b>
You can set logging level with TRAJOPT_LOG_THRESH. Valid values: FATAL ERROR WARN INFO DEBUG TRACE. Defaulting to ERROR
[ INFO] [1675758358.160639990]: Publishing maintained environment on &apos;/tesseract_ros_examples/tesseract_published_environment&apos;
Program: Composite Instruction, Description: Tesseract Composite Instruction
Program: --- Start Instruction, Description: Tesseract Move Instruction
Program: {
Program:   Move Instruction, Move Type: 1, Cart WP: xyz=0.5, -0.2, 0.62
, Description: from_start_plan
Program:   Move Instruction, Move Type: 0, Cart WP: xyz=0.5, 0.3, 0.62
, Description: Tesseract Move Instruction
Program:   Move Instruction, Move Type: 1, State WP: Pos=   -0.4  0.2762       0 -1.3348       0  1.4959       0
, Description: to_end_plan
Program: }
<font color="#CC0000">[ERROR] [1675758358.172219491]: Hit Enter to solve for trajectory.</font>

<font color="#CC0000">[tesseract_ros_examples_basic_cartesian_example_node-3] process has died [pid 1559043, exit code -11, cmd /home/rikka/tesseract_ws/devel/lib/tesseract_ros_examples/tesseract_ros_examples_basic_cartesian_example_node __name:=tesseract_ros_examples_basic_cartesian_example_node __log:=/home/rikka/.ros/log/0b94801a-a6c1-11ed-baa5-0dd8e31ec9e6/tesseract_ros_examples_basic_cartesian_example_node-3.log].</font>
<font color="#CC0000">log file: /home/rikka/.ros/log/0b94801a-a6c1-11ed-baa5-0dd8e31ec9e6/tesseract_ros_examples_basic_cartesian_example_node-3*.log</font>
^C[tesseract_ros_examples_basic_cartesian_example_rviz-4] killing on exit
[virtual_joint_broadcaster_0-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
<b>done</b>
</pre>

Levi-Armstrong, 2023-02-07:
I will test this today

Levi-Armstrong, 2023-02-28:
I tested the demo and they work with my current open PR against tesseract_qt which most likely fixes the issue.

sleichu2, 2023-03-19:
Still not working, using the latest version of the code.

<pre><font color="#4E9A06"><b>rikka@rnc</b></font>:<font color="#3465A4"><b>~/Project/tesseract_master_ws</b></font>$ roslaunch tesseract_ros_examples pick_and_place_example.launch 
... logging to /home/rikka/.ros/log/30942438-c62e-11ed-8fc4-9d31c30c36ea/roslaunch-rnc-1435200.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is &lt;1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
<b>started roslaunch server http://rnc:38225/</b>

SUMMARY
========

PARAMETERS
 * /box_side: 0.2
 * /box_x: 0.15
 * /box_y: 0.4
 * /robot_description: &lt;?xml version=&quot;1....
 * /robot_description_semantic: &lt;?xml version=&quot;1....
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /tesseract_ros_examples_pick_and_place_example_node/file_write_cb: False
 * /tesseract_ros_examples_pick_and_place_example_node/plotting: True
 * /tesseract_ros_examples_pick_and_place_example_node/rviz: True
 * /tesseract_ros_examples_pick_and_place_example_node/steps_per_phase: 10

NODES
  /
    tesseract_ros_examples_pick_and_place_example_node (tesseract_ros_examples/tesseract_ros_examples_pick_and_place_example_node)
    tesseract_ros_examples_pick_and_place_example_rviz (rviz/rviz)

auto-starting new master
<b>process[master]: started with pid [1435210]</b>
<b>ROS_MASTER_URI=http://localhost:11311</b>

<b>setting /run_id to 30942438-c62e-11ed-8fc4-9d31c30c36ea</b>
<b>process[rosout-1]: started with pid [1435225]</b>
started core service [/rosout]
<b>process[tesseract_ros_examples_pick_and_place_example_node-2]: started with pid [1435228]</b>
<b>process[tesseract_ros_examples_pick_and_place_example_rviz-3]: started with pid [1435233]</b>
You can set logging level with TRAJOPT_LOG_THRESH. Valid values: FATAL ERROR WARN INFO DEBUG TRACE. Defaulting to ERROR
[ INFO] [1679213722.467449557]: Publishing maintained environment on &apos;/tesseract_ros_examples/tesseract_published_environment&apos;
Debug:   Created collision object for link box
Debug:   Created collision object for link box
<font color="#CC0000">[ERROR] [1679213722.472347791]: Hit enter key to continue!</font>

Program: Composite Instruction, Description: Tesseract Composite Instruction
Program: {
Program:   Move Instruction, Move Type: 1, State WP: Pos=    0     0     0 -1.57     0     0     0
, Description: Start Instruction
Program:   Move Instruction, Move Type: 1, Cart WP: xyz=0.15, 0.4, 1.127
, Description: From start to pick Approach
Program:   Move Instruction, Move Type: 0, Cart WP: xyz=0.15, 0.4, 0.977
, Description: Pick Approach
Program: }
Debug:   Failed to find or load library: /home/rikka/Project/tesseract_master_ws/devel/lib.so with error: Bad file descriptor
Info:    Pick plan
Debug:   Failed to find or load library: /home/rikka/Project/tesseract_master_ws/devel/lib.so with error: Bad file descriptor
Debug:   Failed to find or load library: /home/rikka/Project/tesseract_master_ws/devel/lib.so with error: Bad file descriptor
Debug:   Failed to find or load library: /home/rikka/Project/tesseract_master_ws/devel/lib.so with error: Bad file descriptor
Debug:   Failed to find or load library: /home/rikka/Project/tesseract_master_ws/devel/lib.so with error: Bad file descriptor
Debug:   Failed to find or load library: /home/rikka/Project/tesseract_master_ws/devel/lib.so with error: Bad file descriptor
Debug:   Failed to find or load library: /home/rikka/Project/tesseract_master_ws/devel/lib.so with error: Bad file descriptor
Debug:   Failed to find or load library: /home/rikka/Project/tesseract_master_ws/devel/lib.so with error: Bad file descriptor
Debug:   Profile &apos;DEFAULT&apos; was not found in namespace &apos;MinLengthTask&apos; for type &apos;N18tesseract_planning16MinLengthProfileE&apos;. Using default if available. Available profiles:
Debug:   Environment, getJointGroup(manipulator) cache miss!
Debug:   Environment, getGroupJointNames(manipulator) cache miss!
Debug:   distances and parents:
Debug:   distance(iiwa_tool0) = 1.306000, parent(iiwa_tool0) = iiwa_link_7
Debug:   distance(iiwa_base) = 0.000000, parent(iiwa_base) = iiwa_base_link
Debug:   distance(world) = 0.810804, parent(world) = workcell_base
Debug:   distance(iiwa_base_link) = 0.000000, parent(iiwa_base_link) = base_link
Debug:   distance(iiwa_link_1) = 0.000000, parent(iiwa_link_1) = iiwa_base_link
Debug:   distance(iiwa_link_2) = 0.360000, parent(iiwa_link_2) = iiwa_link_1
Debug:   distance(iiwa_link_3) = 0.360000, parent(iiwa_link_3) = iiwa_link_2
Debug:   distance(cabinet_1) = 1.350918, parent(cabinet_1) = workcell_base
Debug:   distance(iiwa_link_4) = 0.780000, parent(iiwa_link_4) = iiwa_link_3
Debug:   distance(workcell_base) = 0.039274, parent(workcell_base) = base_link
Debug:   distance(base_link) = 0.000000, parent(base_link) = base_link
Debug:   distance(iiwa_link_5) = 0.780000, parent(iiwa_link_5) = iiwa_link_4
Debug:   distance(iiwa_link_6) = 1.180000, parent(iiwa_link_6) = iiwa_link_5
Debug:   distance(iiwa_link_7) = 1.180000, parent(iiwa_link_7) = iiwa_link_6
Debug:   distance(box) = 0.479189, parent(box) = workcell_base
Debug:   Environment, getJointGroup(manipulator) cache hit!
Debug:   Environment, getKinematicGroup(manipulator, ) cache miss!
Debug:   Environment, getGroupJointNames(manipulator) cache hit!
Debug:   Environment, getKinematicGroup(manipulator, ) cache hit!
[... cut ...]
<font color="#CC0000">[tesseract_ros_examples_pick_and_place_example_node-2] process has died [pid 1435228, exit code -11, cmd /home/rikka/Project/tesseract_master_ws/devel/lib/tesseract_ros_examples/tesseract_ros_examples_pick_and_place_example_node __name:=tesseract_ros_examples_pick_and_place_example_node __log:=/home/rikka/.ros/log/30942438-c62e-11ed-8fc4-9d31c30c36ea/tesseract_ros_examples_pick_and_place_example_node-2.log].</font>
<font color="#CC0000">log file: /home/rikka/.ros/log/30942438-c62e-11ed-8fc4-9d31c30c36ea/tesseract_ros_examples_pick_and_place_example_node-2*.log</font></pre>

Levi-Armstrong, 2023-03-20:
I just ran the example using the latest for all repositories and it does not crash. Can you update the launch file so it launches using GDB and post the back trace?

mr-jun-gsh, 2023-05-22:
I also encountered this problem.
Debug:   Failed to find or load library: /home/rikka/Project/tesseract_master_ws/devel/lib.so with error: Bad file descriptor
I only have lib folder but not lib.so in my ~/tesseract_ws/devel directory, how to solve it ?

mr-jun-gsh, 2023-05-23:
In addition, these errors were found using debug mode：
Debug:   Profile 'cartesian_program' was not found in namespace 'TrajOptMotionPlannerTask' for type 'N18tesseract_planning20TrajOptSolverProfileE'. Using default if available. Available profiles:
Debug:   Profile 'cartesian_program' was not found in namespace 'DiscreteContactCheckTask' for type 'N18tesseract_planning19ContactCheckProfileE'. Using default if available. Available profiles:
Thank you for your reply ^ ^

rjoomen, 2023-05-23:
I cannot reproduce the reported crashes with tessereract_ros:master on ROS Noetic.

To get rid of the profile warnings (not errors), you may want to add the missing profiles like so:
```
auto composite_profile = std::make_shared<TrajOptDefaultCompositeProfile>();
profiles->addProfile<TrajOptCompositeProfile>("TrajOptMotionPlannerTask", "cartesian_program", composite_profile);
auto post_check_profile = std::make_shared<ContactCheckProfile>();
profiles->addProfile<ContactCheckProfile>("DiscreteContactCheckTask", "cartesian_program", post_check_profile);
```

I'm not sure about the library warnings/errors though, they occur also on tesseract_ros2.

rjoomen, 2023-07-21:
I suppose this can be closed.

closed 2023-07-21 by Levi-Armstrong