tesseract-robotics/tesseract_ros #232: Instructions for install

farisnafiah, 2023-12-07:
Hi,

I have been trying to install this for a while. I am using ROS Noetic, and my steps were:

1. git clone https://github.com/tesseract-robotics/tesseract_ros.git
2. wstool init src src/tesseract_ros/dependencies.rosinstall 
3. catkin build --force-cmake

I still keep get this error. Can anyone point out on how to install this package please?

Errors     << ifopt:cmake /home/faris/planner/logs/ifopt/build.cmake.000.log   
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  ifopt_ipopt/cmake/FindIPOPT.cmake:99 (pkg_check_modules)
  ifopt_ipopt/CMakeLists.txt:2 (find_package)

cd /home/faris/planner/build/ifopt; catkin build --get-env ifopt | catkin env -si  /usr/bin/cmake /home/faris/planner/src/ifopt --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/faris/planner/devel -DTESSERACT_ENABLE_TESTING=ON; cd -

...............................................................................
Failed     << ifopt:cmake                                     [ Exited with code 1 ]
Failed    <<< ifopt                                           [ 0.6 seconds ]  
Abandoned <<< osqp_eigen                                      [ Unrelated job failed ]
Abandoned <<< tesseract_command_language                      [ Unrelated job failed ]
Abandoned <<< tesseract_support                               [ Unrelated job failed ]
Abandoned <<< tesseract_geometry                              [ Unrelated job failed ]
Abandoned <<< tesseract_scene_graph                           [ Unrelated job failed ]
Abandoned <<< tesseract_collision                             [ Unrelated job failed ]
Abandoned <<< tesseract_srdf                                  [ Unrelated job failed ]
Abandoned <<< tesseract_time_parameterization                 [ Unrelated job failed ]
Abandoned <<< tesseract_urdf                                  [ Unrelated job failed ]
Abandoned <<< tesseract_state_solver                          [ Unrelated job failed ]
Abandoned <<< tesseract_kinematics                            [ Unrelated job failed ]
Abandoned <<< tesseract_environment                           [ Unrelated job failed ]
Abandoned <<< tesseract_visualization                         [ Unrelated job failed ]
Abandoned <<< trajopt_common                                  [ Unrelated job failed ]
Abandoned <<< trajopt_sco                                     [ Unrelated job failed ]
Abandoned <<< trajopt                                         [ Unrelated job failed ]
Abandoned <<< trajopt_ifopt                                   [ Unrelated job failed ]
Abandoned <<< trajopt_sqp                                     [ Unrelated job failed ]
Abandoned <<< tesseract_motion_planners                       [ Unrelated job failed ]
Abandoned <<< tesseract_task_composer                         [ Unrelated job failed ]
Abandoned <<< tesseract_examples                              [ Unrelated job failed ]
Abandoned <<< tesseract_qt                                    [ Unrelated job failed ]
Abandoned <<< tesseract_rosutils                              [ Unrelated job failed ]
Abandoned <<< tesseract_monitoring                            [ Unrelated job failed ]
Abandoned <<< tesseract_planning_server                       [ Unrelated job failed ]
Abandoned <<< tesseract_qt_ros                                [ Unrelated job failed ]
Abandoned <<< tesseract_ros_examples                          [ Unrelated job failed ]
Abandoned <<< tesseract_rviz                                  [ Unrelated job failed ]
Abandoned <<< vhacd                                           [ Unrelated job failed ]
_______________________________________________________________________________
Warnings   << qpoases:cmake /home/faris/planner/logs/qpoases/build.cmake.000.log
CMake Warning at /home/faris/planner/src/trajopt/trajopt_ext/qpoases/CMakeLists.txt:10 (message):
  No valid qpOASES version found.  Cloning into build directory

marip8, 2023-12-07:
The step you are missing here is the installation of the dependency packages that are distributed on the ROS build farm. `rosdep` should take care of this for you:

```diff
git clone https://github.com/tesseract-robotics/tesseract_ros.git
wstool init src src/tesseract_ros/dependencies.rosinstall
+ rosdep install --from-paths src -iry
catkin build --force-cmake
```

farisnafiah, 2023-12-08:
You're amazing, thanks! It works now.

closed 2023-12-08 by farisnafiah