tesseract-robotics/tesseract_ros #235: I had some problems compiling tesseract_ros

Songkangle2001, 2024-06-04:
My compilation environment is ubuntu20.04 with ros noetic，Below is the problem I encountered：
[image]
[image]
This seems to be a QT problem, I have no way to solve it for now, can you help me, thank you so much！！！

Levi-Armstrong, 2024-06-04:
What are the current tag for all of the tesseract repositories and dependencies?

Levi-Armstrong, 2024-06-04:
The issue should be resolved after this [PR](https://github.com/tesseract-robotics/tesseract_ros/pull/236) is merged.

Jm20010201, 2024-06-05:
Thanks for your reply, now I can compile successfully (it compiles successfully using depencies_unstable.rosinstall, but it still fails when I use the version in dependencies.rosinstall)

Jm20010201, 2024-06-05:
There are some problems when I use **roslaunch tesseract_ros_examples online_planning_example.launch**

[ERROR] [1717582938.070646002]: PluginlibFactory: The plugin for class 'tesseract_rviz/TesseractTrajectory' failed to load.  Error: According to the loaded plugin descriptions the class tesseract_rviz/TesseractTrajectory with base class type rviz::Display does not exist. Declared types are  moveit_rviz_plugin/MotionPlanning moveit_rviz_plugin/PlanningScene moveit_rviz_plugin/RobotState moveit_rviz_plugin/Trajectory rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped tesseract_rviz/TesseractEnvironment tesseract_rviz/TesseractWorkbench
[ERROR] [1717582938.070814547]: PluginlibFactory: The plugin for class 'tesseract_rviz/EnvironmentState' failed to load.  Error: According to the loaded plugin descriptions the class tesseract_rviz/EnvironmentState with base class type rviz::Display does not exist. Declared types are  moveit_rviz_plugin/MotionPlanning moveit_rviz_plugin/PlanningScene moveit_rviz_plugin/RobotState moveit_rviz_plugin/Trajectory rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped tesseract_rviz/TesseractEnvironment tesseract_rviz/TesseractWorkbench

[image]

Levi-Armstrong, 2024-06-05:
It should always build if you are using the `depencies_unstable.rosinstall` but the other should only work if you are pulling the latest tags. Though that is not the case at the moment because I am current working getting a new release out this week.

closed 2024-11-12 by rjoomen