tesseract-robotics/tesseract_ros #239: ros_examples not being animated

zarbod, 2024-07-14:
Hello, I am trying to run some of the tesseract ros examples. When I try to run an example,  it comes up with a correct trajectory but does not animate it. For example, if I run `roslaunch tesseract_ros_examples pick_and_place_example.launch`, I get an error in the terminal that says `[ERROR] [1720975132.052944391]: Hit enter key to continue!`. I have to press enter several times, and at the end rviz has a trajectory generated but the robot doesn't follow through with the trajectory. I have attached a picture to show what happens: 
[image]

What can I do to have the robot actually be animated to follow the path?

Levi-Armstrong, 2024-07-14:
In the GUI there should be a trajectory tab. In that tab you can select the trajectory and then click the play button to simulate.

zarbod, 2024-07-14:
Thank you, that worked. The play button just shows up as a rectangle without a symbol so it's hard to find.

closed 2024-07-14 by zarbod