tesseract-robotics/tesseract_ros #246: Planning failed when an obstacle was created on the path using Command::Ptr addPointCloud()'

sm3304love, 2024-09-03:
Hello. To perform motion planning that avoids obstacles, I modified the [basic_cartesian_example.cpp](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_examples/src/basic_cartesian_example.cpp) code to create a 0.1 x 0.1 x 0.1 cubic obstacle along the end effector’s path.

* Location of obstacles that can be checked in rviz
[image]

The function I used is as follows:
```
Command::Ptr addPointCloud()
{
  // Create octomap and add it to the local environment
  pcl::PointCloud<pcl::PointXYZ> full_cloud;
  double delta = 0.01;
  auto length = static_cast<int>(0.1 / delta);

  // Adjust point cloud generation to center around the specified size
  for (int x = 0; x < length; ++x)
    for (int y = 0; y < length; ++y)
      for (int z = 0; z < length; ++z)
        full_cloud.push_back(pcl::PointXYZ(-0.05F + static_cast<float>(x * delta),
                                           -0.05F + static_cast<float>(y * delta),
                                           -0.05F + static_cast<float>(z * delta)));

  // Add octomap to environment
  Link link_octomap("octomap_attached");

  Visual::Ptr visual = std::make_shared<Visual>();
  visual->origin = Eigen::Isometry3d::Identity();
  visual->origin.translation() = Eigen::Vector3d(0.3, 0, 0.8);
  auto ot = tesseract_geometry::createOctree(full_cloud, 2 * delta, true, true);
  visual->geometry =
      std::make_shared<tesseract_geometry::Octree>(std::move(ot), tesseract_geometry::OctreeSubType::BOX, true, true);
  link_octomap.visual.push_back(visual);

  Collision::Ptr collision = std::make_shared<Collision>();
  collision->origin = visual->origin;
  collision->geometry = visual->geometry;
  link_octomap.collision.push_back(collision);

  Joint joint_octomap("joint_octomap_attached");
  joint_octomap.parent_link_name = "base_link";
  joint_octomap.child_link_name = link_octomap.getName();
  joint_octomap.type = JointType::FIXED;

  return std::make_shared<tesseract_environment::AddLinkCommand>(link_octomap, joint_octomap);
}

```

However, I encountered the following error in the terminal and motion plan failed.

* When using IFOPT
```
Info:    Converged because trust region is tiny
Info:    Not all constraints are satisfied. Increasing constraint penalties for 13
Info:    Not all constraints are satisfied. Increasing constraint penalties for 14
Info:    Not all constraints are satisfied. Increasing constraint penalties for 15
Info:    Not all constraints are satisfied. Increasing constraint penalties for 19
Info:    Not all constraints are satisfied. Increasing constraint penalties for 21
Info:    Not all constraints are satisfied. Increasing constraint penalties for 25
Info:    Not all constraints are satisfied. Increasing constraint penalties for 59
Info:    Not all constraints are satisfied. Increasing constraint penalties for 60
Info:    Not all constraints are satisfied. Increasing constraint penalties for 61
Info:    Not all constraints are satisfied. Increasing constraint penalties for 62
Info:    Not all constraints are satisfied. Increasing constraint penalties for 63
Info:    Not all constraints are satisfied. Increasing constraint penalties for 64
Info:    Not all constraints are satisfied. Increasing constraint penalties for 71
Info:    Not all constraints are satisfied. Increasing constraint penalties for 72
Info:    Not all constraints are satisfied. Increasing constraint penalties for 73
Info:    Not all constraints are satisfied. Increasing constraint penalties for 74
Info:    Not all constraints are satisfied. Increasing constraint penalties for 75
[... 118 lines cut ...]
Info:    Not all constraints are satisfied. Increasing constraint penalties for 72
Info:    Not all constraints are satisfied. Increasing constraint penalties for 73
Info:    Not all constraints are satisfied. Increasing constraint penalties for 74
Info:    Not all constraints are satisfied. Increasing constraint penalties for 75
Info:    Not all constraints are satisfied. Increasing constraint penalties for 76
Info:    Not all constraints are satisfied. Increasing constraint penalties for 77
Info:    Penalty iteration limit, optimization couldn't satisfy all constraints
Info:    TrajOptIfoptMotionPlannerTask motion planning failed (Failed to find valid solution) for process input: Tesseract Composite Instruction
Info:    Planning took 2.701263 seconds.
```

* When not using IFOPT
```
Info:    1 JointVelTermInfo coeffs given. Applying to all 7 joints
Info:    1 JointVelTermInfo targets given. Applying to all 7 joints
[WARN] convex solver failed! set TRAJOPT_LOG_THRESH=DEBUG to see solver output. saving model to /tmp/fail.lp and IIS to /tmp/fail.ilp
[WARN] convex solver failed! set TRAJOPT_LOG_THRESH=DEBUG to see solver output. saving model to /tmp/fail.lp and IIS to /tmp/fail.ilp
[WARN] convex solver failed! set TRAJOPT_LOG_THRESH=DEBUG to see solver output. saving model to /tmp/fail.lp and IIS to /tmp/fail.ilp
[WARN] convex solver failed! set TRAJOPT_LOG_THRESH=DEBUG to see solver output. saving model to /tmp/fail.lp and IIS to /tmp/fail.ilp
[ERROR] The convex solver failed you one too many times.
Info:    TrajOptMotionPlannerTask motion planning failed (Failed to find valid solution: OPT_FAILED) for process input: Tesseract Composite Instruction
Info:    Planning took 8.770013 seconds.

``` 
The only changes made from the example code were the size and position of the obstacle. Is there anything additional that I need to adjust for obstacle avoidance?

Levi-Armstrong, 2024-09-03:
It looks like the Cartesian path goes through the collision object and if that is the case then I would expect it to fail. Since the Cartesian path is a constraint this would make the problem infeasible.

sm3304love, 2024-09-03:
I understand. So, if I want to plan a trajectory that avoids obstacles, would it be better to refer to [pick and place example](https://github.com/tesseract-robotics/tesseract_planning/blob/master/tesseract_examples/src/pick_and_place_example.cpp)?

Levi-Armstrong, 2024-09-04:
I would start with the freepspace example.

sm3304love, 2024-09-04:
I have confirmed through an example that to avoid obstacles, `StateWaypointPoly wp1{StateWaypoint(joint_names, q)};` should be used.

closed 2024-09-04 by sm3304love

Levi-Armstrong, 2024-09-04:
The waypoint is not critical here because you should be able to use JointWaypointPoly, StateWaypointPoly, or CartesianWaypointPoly but you must configure your profile. The profile dictates what constraints and costs are added to the optimizer based on the waypoint type. A good example to review is the freespace glass upright example with uses CartesianWaypointPoly where it disable the translation constraint and only enables the rotational constraint.