tesseract-robotics/tesseract_ros #259: error: no matching function for call to ‘tesseract_planning::ProfileDictionary::addProfile()

CodingHzx, 2024-12-11:
Errors     << tesseract_planning_server:make /home/aiwen/15b/ws_tessss/logs/tesseract_planning_server/build.make.003.log
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp: In member function ‘void tesseract_planning_server::TesseractPlanningServer::Implementation::loadDefaultPlannerProfiles()’:
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:168:97: error: no matching function for call to ‘tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<tesseract_planning::SimplePlannerLVSNoIKPlanProfile>)’
  168 |                          std::make_shared<tesseract_planning::SimplePlannerLVSNoIKPlanProfile>());
      |                                                                                                 ^
In file included from /home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:81:
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note: candidate: ‘template<class ProfileType> void tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<const _Tp>)’
  124 |   void addProfile(const std::string& ns, const std::string& profile_name, std::shared_ptr<const ProfileType> profile)
      |        ^~~~~~~~~~
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note:   template argument deduction/substitution failed:
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:168:97: note:   types ‘const _Tp’ and ‘tesseract_planning::SimplePlannerLVSNoIKPlanProfile’ have incompatible cv-qualifiers
  168 |                          std::make_shared<tesseract_planning::SimplePlannerLVSNoIKPlanProfile>());
      |                                                                                                 ^
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:174:91: error: no matching function for call to ‘tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<tesseract_planning::TrajOptDefaultPlanProfile>)’
  174 |                          std::make_shared<tesseract_planning::TrajOptDefaultPlanProfile>());
      |                                                                                           ^
In file included from /home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:81:
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note: candidate: ‘template<class ProfileType> void tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<const _Tp>)’
  124 |   void addProfile(const std::string& ns, const std::string& profile_name, std::shared_ptr<const ProfileType> profile)
      |        ^~~~~~~~~~
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note:   template argument deduction/substitution failed:
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:174:91: note:   types ‘const _Tp’ and ‘tesseract_planning::TrajOptDefaultPlanProfile’ have incompatible cv-qualifiers
  174 |                          std::make_shared<tesseract_planning::TrajOptDefaultPlanProfile>());
      |                                                                                           ^
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:177:96: error: no matching function for call to ‘tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<tesseract_planning::TrajOptDefaultCompositeProfile>)’
  177 |                          std::make_shared<tesseract_planning::TrajOptDefaultCompositeProfile>());
      |                                                                                                ^
In file included from /home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:81:
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note: candidate: ‘template<class ProfileType> void tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<const _Tp>)’
  124 |   void addProfile(const std::string& ns, const std::string& profile_name, std::shared_ptr<const ProfileType> profile)
      |        ^~~~~~~~~~
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note:   template argument deduction/substitution failed:
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:177:96: note:   types ‘const _Tp’ and ‘tesseract_planning::TrajOptDefaultCompositeProfile’ have incompatible cv-qualifiers
  177 |                          std::make_shared<tesseract_planning::TrajOptDefaultCompositeProfile>());
      |                                                                                                ^
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:180:93: error: no matching function for call to ‘tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<tesseract_planning::TrajOptDefaultSolverProfile>)’
  180 |                          std::make_shared<tesseract_planning::TrajOptDefaultSolverProfile>());
      |                                                                                             ^
In file included from /home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:81:
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note: candidate: ‘template<class ProfileType> void tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<const _Tp>)’
  124 |   void addProfile(const std::string& ns, const std::string& profile_name, std::shared_ptr<const ProfileType> profile)
      |        ^~~~~~~~~~
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note:   template argument deduction/substitution failed:
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:180:93: note:   types ‘const _Tp’ and ‘tesseract_planning::TrajOptDefaultSolverProfile’ have incompatible cv-qualifiers
  180 |                          std::make_shared<tesseract_planning::TrajOptDefaultSolverProfile>());
      |                                                                                             ^
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:187:96: error: no matching function for call to ‘tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<tesseract_planning::TrajOptIfoptDefaultPlanProfile>)’
  187 |                          std::make_shared<tesseract_planning::TrajOptIfoptDefaultPlanProfile>());
      |                                                                                                ^
In file included from /home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:81:
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note: candidate: ‘template<class ProfileType> void tesseract_planning::ProfileDictionary::addProfile(const string&, const string&, std::shared_ptr<const _Tp>)’
  124 |   void addProfile(const std::string& ns, const std::string& profile_name, std::shared_ptr<const ProfileType> profile)
      |        ^~~~~~~~~~
/home/aiwen/15b/ws_tessss/devel/include/tesseract_command_language/profile_dictionary.h:124:8: note:   template argument deduction/substitution failed:
/home/aiwen/15b/ws_tessss/src/tesseract_ros/tesseract_planning_server/src/tesseract_planning_server.cpp:187:96: note:   types ‘const _Tp’ and ‘tesseract_planning::TrajOptIfoptDefaultPlanProfile’ have incompatible cv-qualifiers
  187 |                          std::make_shared<tesseract_planning::TrajOptIfoptDefaultPlanProfile>());
[... cut ...]
[build]   Abandoned: None.                                                                                                                                                                                
[build]   Failed:    1 packages failed.                                                                                                                                                                   
[build] Runtime: 2 minutes and 20.8 seconds total.                                                                                                                                                        
aiwen@aiwen:~/15b/ws_tessss$

Levi-Armstrong, 2024-12-11:
We recently made change. Do you have the latest commits for all of the tesseract repos?

CodingHzx, 2024-12-11:
Thank you. I have installed the latest versions of **tesseract**, **tesseract_ros**, and **trajopt**, but the packages in `dependencies_unstable.rosinstall` are not up to date. I am now compiling with all the latest versions of the required packages.

CodingHzx, 2024-12-11:
[image]
[image]

CodingHzx, 2024-12-11:
Hello, I just compiled using the following package versions on Ubuntu 20.04 with `catkin build`, and encountered the following errors:

CodingHzx, 2024-12-11:
Thank you, the compilation was successfully completed. However, when launching the file using `roslaunch tesseract_ros_examples online_planning_example.launch`, the content displayed in RViz is different from the GIF image. Could you please advise what might be causing this issue? Thank you.
[image]

error：
[image]

CodingHzx, 2024-12-11:
[image]

it is no use,i do not know how to slove it 【https://github.com/tesseract-robotics/trajopt/issues/305 】

Levi-Armstrong, 2024-12-11:
The GUI interface have changed since these were create. When the example pauses go to the trajectory tab and play the trajectory to see it execute.

CodingHzx, 2024-12-12:
[image]
problem resolved, thank you so much for your help!

CodingHzx, 2024-12-12:
problem resolved, thank you so much for your help!
[image]

closed 2024-12-12 by CodingHzx